EdgeTX/edgetx #2285: Label handling with Color Non Touch Radios (like Horus)

ParkerEde, 2022-09-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The new "Label" Feature is difficult to handle with a non touch radio like Horus X10S.
There are many scroll and click actions required to change from one label to another label.
"MDL" longpress, scroll 3 ticks left (to mark the LABEL Box), ENTER (to activate the Label Box), (scroll to select label), ENTER to select label, RTN, scroll to choose Model.
And if you like to change now to another label there is one more step, because you must UNCHECK the previous label
I've tested with selfbuild Commit 58a6239a

### Describe the solution you'd like

It would be very helpfull, if the "Page" Button would get a function, to step the labels one by one (like the previous behavior in 2.7.1)

dlktdr, 2022-09-06:
This has come up before by @pfeerick, and I think your right would be a good addition. I will take this on but before I do I think we should solidify what we want the buttons to do. 

Is everyone okay with,

**Page up** - Selects label above it. If multiple were selected use the highest check mark. (unchecking the others) Wraps around if on top item
**Page down** - Selects label below it. If multiple were selected use the lowest check mark. (unchecking the others) Wraps around if on bottom item

Another thing could probably look at is automatically causing the radio to enter and exit the lblselector instead of having to press ent and return to exit. Not sure how easy this would be, can focus on the buttons first.

rburrow87, 2022-09-06:
Yeah it's a bit cumbersome without touch. I like the idea of page up/down "paging" through the labels like you outlined, @dlktdr.

pfeerick, 2022-09-06:
Agreed, that sounds good. Basically so can just "page" through the labels, like we could categories.

closed 2022-09-09 by pfeerick