EdgeTX/edgetx #2293: Can't scroll to or select models after changing model labels

rburrow87, 2022-09-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After changing the labels on a model, you can't scroll to the models anymore and have to back out of the model selector and go back in to do so.

### Expected Behavior

Should be able to select models after editing model labels.

### Steps To Reproduce

Adjust a model's labels then try scrolling to the models using the scroll wheel.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12, Radiomaster TX16S

### Anything else?

https://www.youtube.com/watch?v=-_TfY6e23Lk

pfeerick, 2022-09-08:
Also on TX16S. Thanks for the video - explains it perfectly :)

dlktdr, 2022-09-08:
Hmm, must be some kind of focus issue with lvgl. Will take a look tonight

raphaelcoeffic, 2022-09-25:
@pfeerick considering #2287 and #2308 have been merged, is this issue still relevant?

pfeerick, 2022-09-25:
Nope, it should have been closed by #2308... Bye bye!

closed 2022-09-25 by pfeerick