EdgeTX/edgetx #2294: some danish translations are to long for 128x64 pixel CLDs

gagarinlg, 2022-09-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

some option names on the "radio setup" and "hardware" pages are longer than the available space.

### Expected Behavior

the texts for the smaller LCDs should be shortened to fit into the available space

### Steps To Reproduce

ise simu/simulator/radio with a small screen and check all pages

### Version

Nightly (Please give date/commit below)

### Transmitter

Other (Please specify below)

### Anything else?

I tested on a BetaFPV Literadio 3 Pro

gagarinlg, 2022-09-07:
@HThuren maybe you can check this?

HThuren, 2022-09-07:
I can, but have only a Taranis 9 plus. Have just checked and changed some to have blanks after the word, but maybe you can provide a picture with the issues.
I have changed some words and uploaded to #2271

gagarinlg, 2022-09-07:
You can use the simulator for this, no need to test on a real radio.
Run cmake for one of the radios with a smaller display and build companion and libsimulator. Then you can test on your PC.

HThuren, 2022-09-07:
Of course, I'll do, Thx

HThuren, 2022-09-07:
Are the right defines -DPCB=X7 -DPCBREV=LR3PRO, I got error from 128x64/model_setup.cpp at struct afhds3, no member runPower and rxFreq
Help !

pfeerick, 2022-09-07:
Yes, 
https://github.com/EdgeTX/edgetx/blob/4e49ac68bed0b19b5fbd063c6d13ee2018627b9f/tools/build-gh.sh#L129

Make sure you delete the `CMakeCache.txt` in your build directory before trying to build for a different radio. Although preferably, just either delete everything in it, or delete the entire folder and re-create it. You end up with left-over files and configuration settings for another transmitter family otherwise, and things go horribly wrong ;) I have a build folder for each different radio type that I build for, so don't have to do that quite as much.

HThuren, 2022-09-07:
Thank you, i try 🙂

HThuren, 2022-09-08:
Great advice, no compiler issues now :-) :-)

gagarinlg, 2022-09-08:
I made me a shell script that runs cmake for a bunch of different radios to different build folders. That way I do not have that issue and can switch quickly between radios without a complete rebuild everytime.

HThuren, 2022-09-08:
Yes, a good solution. I just also now have diffrent build folders, and itt seems to work.

HThuren, 2022-09-09:
Hi @gagarinlg, now changed in  #2301, hope I got them all

pfeerick, 2022-09-19:
Implemented in #2301

closed 2022-09-19 by pfeerick