EdgeTX/edgetx #2298: Internal GPS accessible on AUX1 on TX16S

AJM-DSC, 2022-09-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In the existing software configuration, it is only possible to use Bluetooth **or** the internal GPS. The coexistence of both these additions is not possible because the AUX1port does not have the option to support the interface to the internal GPS module. I have not analyzed the source code, so I cannot assess the reasons for this exclusion on my own. Maybe this option is to be updated in the next firmware release?

### Describe the solution you'd like

Internal GPS support added to AUX1 port.

pfeerick, 2022-09-09:
What version of EdgeTX is this in relation to - as far as I know this was worked since 2.7, not sure how much earlier. Or did you meant AUX2? (if so, this was also raised in https://github.com/EdgeTX/edgetx/issues/2163). As if you do a bluetooth-enabled build of the current `main` branch, you will lose AUX2, not AUX1. And GPS is still an option in AUX1 - I don't have a GPS wired up at the moment so can't confirm it works though.

AJM-DSC, 2022-09-10:
Thanks for your explanation!
My problem with not working internal GPS was that I had no power on the AUX1 of my EACHINE TX16S. The problem was solved by uploading the v2.8 software. there is an option to enable power on AUX1 and AUX2 selectively. Why this is so, I cannot explain. Now that I reinstalled 2.7.1 the GPS worked fine!

pfeerick, 2022-09-11:
That would be because having control over the state of the power pins is a 2.8 feature ;) https://github.com/EdgeTX/edgetx/pull/1848

If that was the only issue, you could have used the VCC and GND pins from AUX 2, and the TX and RX from AUX1 (since that was most likely on due to bluetooth) - not as neat, but still a usable workaround until able to upgrade to 2.8. 

So I take it this is issue is resolved for now?

AJM-DSC, 2022-09-12:
Hi,

 

Problem temporarily solved!

Thx!

 

Andrzej

closed 2022-09-12 by pfeerick