EdgeTX/edgetx #2304: Buttons look different for Model Setup and

mha1, 2022-09-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Button in Radio Setup with Background:
[image]

Same button type in Radio Settings e.g. TOOLS and RADIO SETUP without background:
[image]
[image]

### Expected Behavior

consistent graphical represantation

### Steps To Reproduce

Browse through menus

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

pfeerick, 2022-09-12:
Ah, good point... looks like they are using different colors - can actually see it side by side here with the labels button ;)

[image]
[image]

pfeerick, 2022-09-12:
I suspect the top three "model setup" buttons were done that way to signify active/inactive, but not sure about all the other buttons.

mha1, 2022-09-12:
comparing for example the Internal RF button with the Betaflight setup button in your screen shots I'd say they use different background colors. Same conclusion when looking at my screen shots.

eshifri, 2022-09-12:
Buttons on the setup screen are based on lvgl button; on the tool screen -  lvgl label. (openUI Button and TextButton).

mha1, 2022-09-14:
> Buttons on the setup screen are based on lvgl button; on the tool screen - lvgl label. (openUI Button and TextButton).

Hi, what does that mean in terms of a solution?

eshifri, 2022-09-14:
I do not know, to be honest. :-)
Just a hint to somebody who implemented it.

raphaelcoeffic, 2022-09-15:
These are basically 2 kinds of buttons:
- those that can have 2 states (i.e. status indicator; internal module, trainer, etc...)
- those that have only 1 state (simple push buttons)

If everyone agrees the second type should have the same color as the first in "unactivated state", then let's do it this way.

pfeerick, 2022-09-26:
Well, it's three of the everyone *present*, since he suggested it :-P

gagarinlg, 2022-09-26:
Four 😉

PR #2367 (2022-09-27): fix(color): Unify button colors
Fixes #2304

Requires EdgeTX/libopenui#67
commit d8bcecfc (2022-09-27): fix(color): Unify button colors (#2367)

* Button: change bg color to COLOR_THEME_PRIMARY2

And text color to COLOR_THEME_SECONDARY1

Fixes #2304

* fix: TextButton constuctor calls

Setting the font in the constructor never worked, but did set the text color to black.

* chore: Bump libopenui
commit 1095de20 (2022-09-29): fix(color): Unify button colors (#2367)

* Button: change bg color to COLOR_THEME_PRIMARY2

And text color to COLOR_THEME_SECONDARY1

Fixes #2304

* fix: TextButton constuctor calls

Setting the font in the constructor never worked, but did set the text color to black.

* chore: Bump libopenui
closed 2022-09-27 by pfeerick