EdgeTX/edgetx #2307: 6P multi pis switch doesn’t work on 2.7.1

macfly1202, 2022-09-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Even if 6P switch on TX16S is calibrated, only 6P1 and 6P2 positions are working. Because the 6 position is use by most of ardupilot community, currently we ca ‘t use edge rx without this 6 positions switch ok. 

### Expected Behavior

Each position from 6P1 to 6P6 must work like on open TX.

### Steps To Reproduce

Try to use 6 positon switch and you will see only 6S p1 and p2 are working. I have check in configuration/calibration menu of radio and a ‘multi pos’ switch is selects. 

### Version

2.7.1

### Transmitter

Radiomaster TX16S

dlktdr, 2022-09-12:
I have 2.7.1 on my TX16s and all 6 positions work okay. I also use it for Ardupilot for various flight modes.

When your calibrating are you starting with the 6P on P3 or P4 when it asks to you to center your sticks? I always use P3, then when it asks to put the sticks to their min/max press P1 then P6.

pfeerick, 2022-09-12:
2.7.1 should not have any issues with the pots - the only times there have been any issues reported it has been around the 6POS not being calibrated, or a hardware issue (i.e. the ribbon cable needed reseating). I've not had any issues on my primary radio, or with my testing ones running 2.7.1 through to current 2.8 builds. 

As Cliff said, when you do the stick (analogs, actually - since pots and sliders aren't sticks!), you also need to do the 6POS because it is really an analog input. If you don't, you will get weird things like only one or two buttons working. 

Working 2.7.1 6POS => https://youtu.be/gWeLQH8tdzE

Calibration => https://youtu.be/QoK-icJEaW0

macfly1202, 2022-09-12:
Hi Peter, 

I carefully read your message and redo the process. And it works ! 
What people need to understand in calibration step is that you must click to 1 then 6 and not 1, 2 , 3 ... it doesn't woork at all and be counter-intuitive. Because clicking from 1 to 2 display no reaction , I was tthinking it's like a standard switch. I was wrong. 

Thanks you so much taking the time to make a so clear video. 
Now my TX16s with Yaapu script with Google maps works nicely with my arduplane models. I will test this extensively soon.  
 
KR,

Jean-Marie

ps: you can close this issue

closed 2022-09-12 by rotorman