EdgeTX/edgetx #2311: [NV14] ExpressLRS Team900 - constantly power switching

nightflasher, 2022-09-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With ExpressLRS 3.0.0-RC2 I also have the ELRSwidget (should be pre-installed on edgeTX!) to get some information on the link, as well as a screen with elrs telemetry and model telemetry.
In ELRS 3.0.0-RC2 the 868 MHZ module (ttgo_v2) was constantly switching through the power values, even the dynamic switching isn't set (see the video in https://github.com/ExpressLRS/ExpressLRS/issues/1686#issuecomment-1223574718). With the ELRS release of 3.0.0, the module isn't switching wildly - but the TPWR and the power in the ELRSwidget are still showing, that the module is switching - but it doesn't (at least the oled shows stable power). 

### Expected Behavior

Showing the selected and hopefully used transmitting power. 

### Steps To Reproduce

Flash ttgo_v2 with ExpressLRS 3.0.0 and set up telemetry screen.

### Version

2.7.1

### Transmitter

FlySky NV14

gagarinlg, 2022-09-12:
ExpressLRS is a separate project with different people and it's own release cycle. So always having the correct Lua script is essentially impossible. The ExpressLRS configurator doey show a link to the Lua script.
The described behavior is caused by the TX and possibly RX firmware and can not be created by EdgeTX.
The NV14 is know to have a dodgy power supply for external modules. This might add to your issues.

nightflasher, 2022-09-13:
Thanks, but imma former dev to elrs too.
I was talking about a widget, doesn't change a lot there. 
I've done every mod on the NV14 that needed to be done and it's working great now.
But here is something that was addressed to ExpressLRS and is now fixed, but the NV14 still shows power switching cycles, but the module doesn't switch. So this could be an edgeTX problem.

pfeerick, 2022-09-13:
I have a 900Mhz module (the discontinued Happymodel one 😓 ) that I can dig out and flash v3 onto... and see if the same happens for me with TX16S and NV14... at least will be another point of reference.

pfeerick, 2022-09-15:
Well, I got my HM ES915TX and HM ES900RX back into service and on ELRS v3.0.0, nad when set to dynamic power it is rock steady on the NV14... walked outside to make it want to boost the power above 10mw... took more work than I remembered it needing but even then it bumped to 25mw, and then ramped back down again as I started going back in, and has stayed steady for the last 5-6 minutes. Mind you, this is with a nightly build from a few days ago, and there have been a few things done that would affect the NV14 so, it may be worth doing a comparison test to see if it still does it there for you.

nightflasher, 2022-09-15:
Here you go
https://vimeo.com/749887466

pfeerick, 2022-09-15:
Hm... interesting... your telemetry is also indicating transmit power of the module is flapping about all over the place... about the same frequency it was when the OLED was showing it... Can you actually pin down a fix that targeted this issue? As I can only see a commit that doesn't affect this hardware, and other commits which were in RC2... so I half-feel it wasn't resolved at all, but is still lurking. 

https://streamable.com/93fndu

nightflasher, 2022-09-15:
Hmm...that's interesting.
A wild guess: NV14 mainboard revision depended?
I have to test some more 868MHZ modules (r9m & lite).

nightflasher, 2022-09-15:
i've tested my ttgo_v1 and it works just fine - BUT it's 10mW only...so i've tested frksy r9m lite nad it stays rock solid as you would say. maybe it's the ttgo_v2, so i need to get hold of another one.

let's set this issue on hold!?

closed 2023-05-26 by gagarinlg