EdgeTX/edgetx #2315: unify the color of the top widgets

tonims1, 2022-09-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I suggest unifying the color of the top widgets so that when you change the theme you don't have to change the color of all models by model

### Describe the solution you'd like

When changing a theme that has a white top bar, if you have all the widgets in white, they are not visible, the same if the bar is black and the color of the widgets is black.
If the colors were unified, only changing once, it would not have to be changed in all the models.

### Describe alternatives you've considered

if they could be included in the theme editor it would be fantastic

pfeerick, 2022-09-14:
That is really contingent on another issue - https://github.com/EdgeTX/edgetx/issues/1840 - as we need more options in widgets to be able to do that. Because colors are a option for widget, we can't make the assumption that you want the widgets to be the theme color. So need to be able to add an option to the widget that lets you indicate that case - i.e. like with the "Models" widget. Now, having said that, almost all of the built-in widgets actually have space for one more option. IMO it would be better for all of the built-in widgets to have that option, rather than have one or two odd ones out... or at least that is my excuse for only adding it to the Models widget while I was making other changes to it! 😆 . However, that will only account for built-in widgets - it is up to the Lua widget authors what their widgets do - having said that - it is an easily editable file in most cases. Also, there is now the ability to replace/override built-in widgets with Lua widgets (#2208), allowing for this sort of thing to happen that way.

tonims1, 2022-09-14:
Yes . integrated widgets such as value, transmitter battery voltage or telemetry values. Perhaps the easiest thing is that they change according to the selected theme.

pfeerick, 2022-09-14:
And what happens to the users who *don't* want them to change, and had them set to specific colours? 🤔

tonims1, 2022-09-14:
maybe put a selector same as model widget, use color theme

pfeerick, 2022-09-14:
Which brings us back to needing more options, as we can only have a maximum of five options for widgets atm, and I know of at least one if not more of the built-in widgets that already has five options so doesn't have room for any more. 

That's why the options at the moment are probably:
1) add it to the ones that can still have room (not consistent, but such is life)
2) wait for widgets to have more options, and then add it to all them (are we there yet? 😆)
3) when we add model-specific UI, including per model themes, maybe add an option there that *all* built-in widgets should honor theme color... 
4) use Lua widgets that override the built in widgets and make the desired changes there

All but 4 involve a wait for 2.9 or later... the 4th one is possible in the nightlies now, and thus also 2.8 when it releases.