EdgeTX/edgetx #2322: Trim travel wrong direction

Eldenroot, 2022-09-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Opening this issue because it is still present also in edgetx.

if you set "weight" in Input Tab to negativ value trim travel always positiv

### Expected Behavior

https://github.com/opentx/opentx/issues/8950#issuecomment-1246373140

If a stick is inverted in the inputs,
the corresponding trimmer is also inverted.
This solved the problem.

### Steps To Reproduce

Check above.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

### Anything else?

Maybe could be fixed by this?
https://github.com/opentx/opentx/issues/8950#issuecomment-1246373140

pfeerick, 2022-09-16:
Nope, because that is code for Ethos - that was Bertrands point! 😆 I do agree that if you reverse the put direction, it makes sense that the trim would work accordingly though.

Eldenroot, 2022-09-16:
OK, well :)

I hope it is not difficult to fix :) because now it is quite confusing:(

closed 2026-11-05 by pfeerick