EdgeTX/edgetx #2331: The value of GV1 has set itself to the value 0 in FM1 resp. FM2

ParkerEde, 2022-09-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The value of the varibal GV1 has set itself to the value 0. This has happened to me twice in the meantime, once GV1 FM1 and another time GV1 FM2 was affected. Unfortunately I could not identify a way to reproduce it. Once I was in the Theme Editor, changed a few colors there, then into the Model Editor to see how colors work. Among other things also in the flight modes. There I used the switch to toggle the flight modes to test the highlighting and the scroll wheel to test the focus. But as I said it seems to happen only sporadically. I have a Horus X10S Express and a 2.8.0 selfbuild (e4fc166e).

Translated with www.DeepL.com/Translator (free version)

### Expected Behavior

the values of the GV's must never change by themselves

### Steps To Reproduce

Unfortunately I could not identify a way to reproduce it.
I guess once you navigate through other Modelleditor pages multiple times (without changing anything) it happens sporadically. I'll keep trying to provoke it though and will report back as soon as I find out more.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.8.0 selfbuild (e4fc166e).

raphaelcoeffic, 2022-09-20:
Is it thinkable that you tried to delete a theme after you've edited it? I've found an error with that while testing (see e8f9f646e0579). Or did you experience it even if you did not use anything related to the theme editor?

ParkerEde, 2022-09-20:
In the two cases that have occurred in my case, I was also in the Theme Editor. However, I did not try to delete the theme or delete it by mistake. I merely edited it. Can you explain to me what the connection is between the Theme Editor and the global variables? Then I could test it a little better.

raphaelcoeffic, 2022-09-21:
> In the two cases that have occurred in my case, I was also in the Theme Editor. However, I did not try to delete the theme or delete it by mistake. I merely edited it. Can you explain to me what the [...]

There is none, in fact. However I found some incorrect memory access in the Theme editor that could have explained these random changes.

ParkerEde, 2022-09-21:
Now I have created a firmware bin based on (8b7e8c30) and manually added the line 
MainWindow::instance()->run(false);
in the file 
[radio/src/gui/colorlcd/fullscreen_dialog.cpp].

Next weekend I'm going to an indoor event. We'll see what happens.

pfeerick, 2022-09-21:
Did you need custom build flags for something? As those fixes were in yesterday's nightly ;)

ParkerEde, 2022-09-22:
no, but when I wanted to build it yesterday, the change was not yet included in the Main Branch. I cloned and built it again today. It's now 8b7e8c30 and I'll report back.

ParkerEde, 2022-09-25:
I have not been able to recreate the problem so far. I will keep an eye on it, but I think we can close the ticket for now. If something else comes up, I would open it again.

closed 2022-09-25 by raphaelcoeffic