EdgeTX/edgetx #2334: Going from open tx: partial throttle range

macfly1202, 2022-09-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I backup my open tx model. When I import them into edge tx, everything seems ok at first sight but that’s not the case. Throttle has a partial range despite setup from -100 to 100%. I manually created a model on edge tx and I have full throttle range on this model (model16). On my imported model (s800 2022) throttle go from -100 to +3% despite a max range at 100%. 

### Expected Behavior

Throttle range must go from -100 to +100% 

### Steps To Reproduce

See the video 
https://youtu.be/nN7PY5OygF4

### Version

2.7.1

### Transmitter

Radiomaster TX16S

elecpower, 2022-09-19:
Try recalibration and see if it helps

macfly1202, 2022-09-19:
@elecpower nice advice but I already done it multiple time. Creating new model on EdgeTx give correct range for throtlle. Only older plane have this. I'm trying to find a workaround but currently that's means no plane can't be use because throttle range are not correct.

pfeerick, 2022-09-19:
Do you have the original bin files still? If so, can you zip up the RADIO and MODEL folders so we can get some idea where things went wrong.

macfly1202, 2022-09-19:
I found the problem. Importing model also imported the Trim that was not in neutral and set to exponential. 
Again, another issue between the chair and the keyboard 😁
Thanks for your help. I nearly finish to check all migrated elemented from Open TX, with success ! great job with Edge TX

pfeerick, 2022-09-19:
🤣 As long as it works, and we have another happy customer, all good! :)

closed 2022-09-19 by pfeerick