EdgeTX/edgetx #2338: 'Slow servo' option is not linear and one enhancement needed

Billy60, 2022-09-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

1) the option to slow the servos acts non linearly: it starts indeed slow but then accelerates a lot. it is more like a kind of exponential movement.
Could it be 'slow linear' instad of 'slow expo' ?
2) in the functions, there is "a=x". unfortunately the hadware is not always precise enough. Could it be possible to have also "a more-or-less-equal to x", just as there is in Otx ?

Thanks !

### Describe the solution you'd like

It should be a slow and linear movement, that is with a linear slower speed of the servo on all its travel.
A "a almost-equal-to x " option should be in the Functions list choice.

pfeerick, 2022-09-21:
Regarding no. 2, `a~x` is still there - on colorlcd it just showed up as `ab` but that was fixed somewhere along the way in 2.8.

eshifri, 2022-09-21:
Regarding #1 can you please clarify: angular speed is not linear or linear speed is not linear? (The second one would be expected if the first one is true. :-) )

Billy60, 2022-09-22:
Good morning
Well, both are not linear. The servo starts slowly and then accelerates a lot. I would say its speed is exponential and not constant. So the angular speed is not linear. The servo should travel at the same speed, slower than usual as defined.

By the way, is there a way to Copy the curves ?

Thanks !

eshifri, 2022-09-22:
On the channel monitor the speed looks constant. 
You do not have a curve associated with this input, do you?

The curve can be copied using a text editor on the model.yml file. 
Maybe there is a way to do it in Companion, but I do not know it.

Billy60, 2022-09-22:
No curve associated in my case.Could it be possible to copy the curves like it is in Ethos : ‘copy’  and it is done, directly in the Curves menu ? Envoyé à partir de Courrier pour Windows De : eshifriEnvoyé le :jeudi 22 septembre 2022 08:05À : EdgeTX/edgetxCc : Billy60; AuthorObjet :Re: [EdgeTX/edgetx] 'Slow servo' option is not linear and one enhancement needed (Issue #2338) On the channel monitor the speed looks constant.You do not have a curve associated with this input, do you?The curve can be copied using a text editor on the model.yml file.Maybe there is a way to do it in Companion, but I do not know it.

pfeerick, 2022-09-22:
Please stick to the topic at hand. If you want to request features, etc, please raise that as a separate issue. 

And yes, you can copy and move curves in Companion.