EdgeTX/edgetx #2339: Nightly 2.8 - Yaapu LUA screen slightly off-center

robustini, 2022-09-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

As you can see from the image I posted, Yaapu LUA's screen is slightly off-center to the right, as can be seen from the vertical white band on the left.
It is absolutely a small thing but it would be better to bring it back to center.
I believe that the same imperfection also exists with other LUAs.

### Expected Behavior

That the image was centered

### Steps To Reproduce

1. Flash the latest nightly build
2 . Install Yaapu LUA script

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

[image]

pfeerick, 2022-09-21:
Reproducible with Yaapu telemetry widget 1.9.6-dev-4265d2f. The red outline frame which is shown if there is no telemetry is also off-screen by one column.

stronnag, 2022-09-21:
Save affect is seen with the INAV telemetry widget.

pfeerick, 2022-09-25:
I noticed an interesting behaviour - if you make the widget enter full-screen (interactive) mode... it then aligns properly... 🤔 

i.e. on the left it's just "normal" fullscreen, and you can see that 1px sliver on the left. I then press 'n held, choose "full screen" and now on the right it's correctly positioned as you can see the full red outline for 'no telem'

[image]

stronnag, 2022-09-25:
Me too!!!!
INAV widget behaves the same way (looks correct on manual Full Screen).

robustini, 2022-09-25:
@yaapu what do you think?

raphaelcoeffic, 2022-09-25:
Smells like LVGL style border or similar.

banyaszg, 2022-09-26:
It is reproducible with a simple full screen widget which just fills the area with a color. The left line is the background image.

PR #2374 (2022-09-27): fix(color): Remove Main View - Main Zone left side offset
Fixes #2339
closed 2022-09-27 by pfeerick