EdgeTX/edgetx #2342: EdgeTX:Companion:Telemetry:Load and Write Modeldata changes subid to +1

frank-hubgit, 2022-09-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Hello, 
I use a Frsky X12S with ISRM-Modul, Edgetx 2.7.1 and Companion. 
It is possible to load all the models from the x12s into companion. Changing data is possible too. Everything seems to work fine. But when I write the data back to the x12s the telemetry data is wrong.  
Logical switches and special functions do not deliver the correct data. 
I use mostly the G-RX8 receiver including the telemetry.
As I saw all the subids from all models are added with 1.  
subid 16 becomes 17 and so on.  
Do I have to do any configuration-change at companion? Is this issue known?  
Thanks for your hints. 
Kind regards 
Frank
--
  |  
 

### Expected Behavior

subids should be unchanged if models are loaded into companion and saved back to the x12s.

### Steps To Reproduce

each time models are loaded into companion and saved back to the transmitter.

### Version

2.7.1

### Transmitter

FrSky X12

frank-hubgit, 2022-09-21:
Hopefully this is the right place to commit this point...

pfeerick, 2022-09-22:
So making changes on the radio is fine? Is it only when Companion is used to read and write? Does it only do it the one time, or does it keep incrementing each time you read and write?

It would probably be best if you can zip up your MODELS and RADIO folder, and upload that here. Also, if you just read the radio, and write to it, does it mess the settings up then?

frank-hubgit, 2022-09-22:
Looking at the radio the subid ist 16. After loading into companion it shows 17.
After writing back to radio it shows 16.
It does not add up to 99.
Only the number is one higher when I look at companion.

Why did I post this?
Since using edgetx I have trouble with telemetry. I have some gliders with frsky g-rx8 receiver. After writing back from companion into the radio the telemetry is wrong. I always have to delete and research some variables like alt or v-speed. Example: After using switch sh the altitude, the ESCV and the ESCC shall be said, but it says a other telemetry.
After redefining it works until it was loaded into companion and written back...
[frsky-x12s-220922.etx.zip](https://github.com/EdgeTX/edgetx/files/9622568/frsky-x12s-220922.etx.zip)
actual I am using the "Takeo".

pfeerick, 2022-09-22:
That sounds promising - It could simply be that Companion is not taking into account index starts from 0, instead of 1 - in other words, a simple visual mistake (they are the most confusing!). Thanks for attaching those files.

frank-hubgit, 2022-09-22:
yes for companion. it seems that someone starts with 0 and the other one with 1.
Does it repair my problem with the changing telemetry?
From my side at the moment I have to research mostly all telemetry in order to work correct...