EdgeTX/edgetx #2343: Companion not work with ext. XJT Lite in Zorro

meyer1111, 2022-09-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

XJT Lite module works fine with radiomaster zorro. But not with companion (tested with 2.7.0 and 2.7.1).
error message:
"Warning: The internal module protocol **FrSky XJT lite (D16)** is incompatible with the hardware internal module **MULTI** and has been set to **OFF**!!

But, i will not use internal module. External module works fine. I can bind FrSky Receiver without problems.

### Expected Behavior

companion shall **not replace** "XJT Lite D16" with "FrSky XJT (D16)".

### Steps To Reproduce

1. edit model with companion

### Version

2.7.1

### Transmitter

Radiomaster Zorro

### Anything else?

[image]
[image]
[image]
[image]

pfeerick, 2022-09-22:
I can reproduce this on my ELRS Zorro with both 2.7 and 2.8-dev Companion when the default Int. module is set to CRSF and radio settings as read from the radio agree that int. is CRSF. XJTLite is not even an option for internal anyway, so Companion seems to be mis-reading something. It seems somewhat specific to the XJTLite setting as setting Ext. to GHST was processed fine. 

[image]

[zorro-xjtlite.etx.zip](https://github.com/EdgeTX/edgetx/files/9624971/zorro-xjtlite.etx.zip)

closed 2022-09-22 by pfeerick

pfeerick, 2022-09-22:
This should be fixed for 2.8, which will hopefully be going into RC soon. You should be able to get away with grabbing a 2.8-dev build of companion still though in about two hours if you need it now ;)

Companion (main branch): 
- Windows =>  https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain
- Mac => https://github.com/EdgeTX/edgetx/actions/workflows/macosx_cpn.yml?query=branch%3Amain
- Linux => https://github.com/EdgeTX/edgetx/actions/workflows/linux_cpn.yml?query=branch%3Amain

meyer1111, 2022-09-22:
Many thanks for your help!

pfeerick, 2022-09-22:
Many thanks for reporting it! 😄

commit 40149c48 (2022-09-22): fix(cpn): Zorro, not T-Lite talks to Lite modules
Also, BetaFPV LR3Pro doesn't talk to Lite modules
Fixes #2343