EdgeTX/edgetx #2349: Getting low RSSI while using Flysky rxs on 2.8

Mikey1968, 2022-09-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I was on 2.7 everything works great, when I went to 2.8 Nightly and flew my planes I was getting low rssi warnings that I never had before 

### Expected Behavior

Hopefully you guys fixed the problem 

### Steps To Reproduce

NA

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

raphaelcoeffic, 2022-09-25:
@Mikey1968 we will need a bit more detail ;-)

Are these assumptions correct?
- Multi-protocol Module (MPM)
- AFHDS2A
- Original ADFHS2A receiver? (which one? I have only very few)

Could you give us more details on the specific of what you consider low RSSI values? How where they before, how does it look like now?

Mikey1968, 2022-09-26:
I have the Radiomaster TX16s Mk2 using AFHDS2A using the FS-iA6B rx .  Before when using 2.7 everything works great no rssi while flying . When I updated to 2.8 Nightly everything was working great until I did a flight and noticed at close flying I was getting rssi warnings. So decided to land . Wasnt trusting it so went back to 2.7 .  Might be nothing and just a glitch . 
Im going to download the new 2.8 RC1 and see if it happens again

Mikey1968, 2022-09-26:
also another thing I noticed , when you pick a track , you used to be able to scroll backward s from X to A to pick a sound. Put the version I was using you could only scroll a-z

pfeerick, 2022-09-27:
The actual RSSI is unlikely to change, since everything RF is the job of the RF module, in this case the internal MPM. So you shouldn't have any changes there - only potentially how ETX is interpreting the RSSI/or the warning value being lower (higher?). 

And yes, all the lists are one way now, but that is a *different* issue - whether that will change I don't know - the velocity tweaks make it so you can go from one end of the list to the other with nearly one fast turn of the rotary encoder. ;)