EdgeTX/edgetx #2352: Tapping in "Hardware" on a TX16S causes Emergency Mode

ChristianPuettner, 2022-09-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When tapping on the touchscreen in "Hardware"-menu on a fresh to EdgeTX migrated Radiomaster TX16S causes Emergency Mode.

### Expected Behavior

It should not go into Emergency Mode.

### Steps To Reproduce

- On the TX16X, go to the "Hardware" menu
- Tap or double tap on the center of the touchscreen 

This video shows the issue: [EdgeTX issue](https://youtu.be/ggBjjDja4qw)

### Version

2.7.1

### Transmitter

Radiomaster TX16S

pfeerick, 2022-09-25:
Ouch... that's not good. I can't reproduce this on mine. But you mention "migrated", so this is after having imported some settings from OpenTX? Can we get a zip of your RADIO and MODELS folder then, as I'm suspecting there is something in the radio configuration that is causing this - maybe a settings conversion went badly?

eshifri, 2022-09-25:
I also cannot reproduce it neither on TX16S nor NV14 in both 2.7.1 and 2.8.
And it looks like the screen was not completely painted in the video.

ChristianPuettner, 2022-09-25:
I found my mistake... I didn't flash the Bootloader.

FIRMWARE.TXT:
____________________________________________________________
EdgeTX Firmware for tx16s

FWVERSION  edgetx-tx16s-2.7.1 (a55aff02)
DATE       2022-05-09
TIME       04:02:03
BOOTVER    opentx-tx16s-2.3.11-otx (4328055e)
____________________________________________________________

After flashing the Bootloader (and the Firmware again) the issue is gone.

closed 2022-09-25 by ChristianPuettner

pfeerick, 2022-09-25:
Very weird - I wouldn't have *expected* the bootloader not having been updated to cause a problem, but weirder things have happen. Glad it's working properly for you! :)