EdgeTX/edgetx #2357: More german translations

ParkerEde, 2022-09-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have made  many german translations for 2.8.0 but I have no permission, to publish the branch and create a pull request.

### Describe the solution you'd like

@pfeerick  Can you help me to get the permissions?

raphaelcoeffic, 2022-09-25:
@ParkerEde anyone can create a PR. What you need is first fork the project, then add the branch to your fork. Then you can make a PR against EdgeTX. 

Here is some help from GitHub docs: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

closed 2022-09-25 by raphaelcoeffic

ParkerEde, 2022-09-25:
perfect. Thank you. PR is now created.