EdgeTX/edgetx #2358: audiowarning: "telemtry lost" comes twice when changing model

ParkerEde, 2022-09-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

when i fly a model, end the flight and disconnect the battery, i get a "telemetry lost" message. Good so far. But now when I select another model, I get the same message again. This should not be the case.

### Expected Behavior

when the model is changed, the message should only come when the battery is disconnected, but not when the model is changed afterwards.

### Steps To Reproduce

1. Open Preset for Model1 and fly
2. After landing unplug the battery
3. You got the message "telemtry lost"
4. Change the Model preset to Model2 (now the message "telemetry lost"comes again)

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.8.0 selfbuild https://github.com/EdgeTX/edgetx/commit/8b7e8c30170bc10d82a3c3618e54cac73be046ce

eshifri, 2022-09-25:
Do you reset telemetry (Special Function) after the battery is disconnected?

ParkerEde, 2022-09-25:
Yes exactly, you are right. I had forgotten about this step.
I've update the original Post. Thank you.

closed 2022-11-04 by ParkerEde