EdgeTX/edgetx #2361: Calibration screen improvements

pfeerick, 2022-09-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The current calibration process is quite annoying - it doesn't do some things that it could quite easily, and doesn't always tell you *everything* you need to know. While I would also like to attack the general onboarding experience as part of this (partly due to the radio setup corruption PR, I think that needs to be done separately. 

### Describe the solution you'd like

Generally, it should be 
- touch and hard key friendly
- tell you to calibrate the 6POS (until/unless that requirement is removed)

When doing the calibration of analog inputs, it should, in the case of a first run calibration:
- tell you to move the gimbals in a specific order, so that the stick mode can be auto detected. 
- prompt for the default channel order

When doing the calibration of analog inputs, it should, in the case of a re-calibration:
- skip any start screen - i.e. jump right to centering
- doing next/ENT on the final screen should not start the calibration process off again

Eldenroot, 2022-09-30:
Especially 6pos switch on tx16s

raphaelcoeffic, 2022-10-01:
> tell you to calibrate the 6POS

that requirement really needs to go away! @rotorman we need data 🤗

rotorman, 2022-10-01:
> that requirement really needs to go away! @rotorman we need data 🤗

Thanks to raw unfiltered analog output display in EdgeTX, we could easily use the community to gather some sample raw ADC values from radios that have analog 6-pos (TX16S, T16, T18).
I am myself not near a radio with 6-pos for next 2 weeks, thus cannot help out quickly with some values at the moment.

gagarinlg, 2022-10-01:
on mine:
TX16S MK2 40,281,570,860,1150,1432
TX16S MK1 0,283,599,907,1223,1528
TX16S MK1 0,283,594,907.1219.1531

O-to-the-o, 2022-10-01:
Tx16s mk1: 0, 292, 603, 920, 1230, 1544
All a bit fluctuate +/- 1

rburrow87, 2022-10-01:
RM TX16S MK2: 0, 263, 550, 847, 1116, 1402
Eachine TX16S: 0, 248, 539, 836, 1106, 1397
X12S 6 pos knob: 502, 814, 1018, 1357, 2040, 4091

banyaszg, 2022-10-01:
RM TX16S mk2: 0, 256, 540, 830, 1096, 1380

jtaylor2, 2022-10-01:
RM TX16s MK1 0, 286, 602, 921, 1238, 1550

mha1, 2022-10-02:
RM TX16s MK1: 0, 259, 548, 838, 1103, 1396

pfeerick, 2022-10-03:
RM TX16S MK1, 0, 293, 610, 928, 1245, 1561
RM TX16S MK1, 0, 281, 586, 899, 1205, 1519
RM TX16S MK2, 0, 264, 553, 844, 1112, 1403
RM TX16S MK2, 0, 253, 544, 836, 1103, 1393

pfeerick, 2022-10-03:
These are the results so far : https://docs.google.com/spreadsheets/d/1p3J7AUZ4oDhM3nNM5vfk0bHZHL13SXwb59qaR2ldnQg/edit?usp=sharing

rotorman, 2022-10-03:
From the data so far (we are still missing T16, T18 and X12S is also represented by a single specimen), we see that a uniform classification borders across radios is not possible, but on TX16S at least skipping the end-user calibration is possible - there are no overlapping regions:
[image]

ulfhedlund, 2022-10-03:
TX16S
1:  -100 0
2: -60 409
3: -20 819
4: 19 1228
5: 59 1638
6: 99 2047

gagarinlg, 2022-10-03:
> On TX16S: Pos 1 - a value equal or lower to 140 Pos 2 - a value between 141 and 429 Pos 3 - a value between 430 and 729 Pos 4 - a value between 730 and 1029 Pos 5 - a value between 1030 and 1327 Pos 6 - a value equal or higher than 1328

And with @ulfhedlund s values, that won't work anymore :(

pfeerick, 2022-10-03:
@ulfhedlund Is that from the third tab - the "unfiltered raw analogs" - as that looks more like the values from the first tab - calibrated analogs

gjhaynes, 2022-10-05:
TX16S Mk2
0, 259, 549, 833, 1101, 1388

ulfhedlund, 2022-10-11:
You are of course absolutely right,  sorry.The correct values should be, from 1 to 6:0, 294, 612, 932, 1251, 1573

pfeerick, 2022-10-11:
Phew! you had us worried there! Thanks for the update :)

rotorman, 2022-10-28:
I tested my TX16Ss as well, here the combined results with the detection borders of all of our data (looking good - all 6 position over all samples can be separated correctly):
[image]

The updated classification borders for TX16S (Gen.1, mkII and Eachine):
Position 1 - 138 and below
Position 2 - from 139 to 427
Position 3 - from 428 to 719
Position 4 - from 720 to 1019
Position 5 - from 1020 to 1326
Position 6 - 1327 and above

edernucci, 2022-10-28:
**TX16S MK1**
0 292 612 934 1256 1577

andytransonic, 2022-10-29:
RM TX16S Mk 1- 0, 255, 546, 836, 1099, 1387

rotorman, 2022-10-29:
Thanks! With these two TX16S additions, the classification borders remain exactly as 3 posts above.

Eldenroot, 2022-10-29:
Guys, I do not have the transmitter at home right now... anyway I have Radiomaster TX16S Mk II. I will try to send the values in 1-2 weeks when I get to the transmitter.

rotorman, 2022-10-29:
Thanks!
We really need data from **Jumper T16, Jumper T18, FrSky X10 and FrSky X12S users** (currently only a single X12S data record from rburrow87 above in addition to numerous TX16S samples).

Eldenroot, 2022-10-29:
EdgeTX users just love quality and open systems :) Thats why we have so many TX16S samples :)

pulpfiction15, 2022-10-29:
Jumper T18

-100 0
-60  305
-20 614
19 923
59 1225
99 1555

1Sven, 2022-10-30:
For X12s is the hardwarescreen not useble yet. Only emti values. The process of calibratin are not working...

rotorman, 2022-10-30:
Are you sure? Seems to work fine in simulator though (channel 06):

[image]

raphaelcoeffic, 2022-10-30:
@rotorman we have a live session planned for next Sunday with @1Sven

pfeerick, 2022-10-31:
@rotorman Make that a single data point for X10 and X12S - per the spreadsheet linked in the first post comment. 

@1Sven You should still be able to get the raw values - as this is outside the calibration process. The whole goal here is to eliminate the need for calibration of the 6POS switch if sufficient samples can be collected, as then we can determine a range for each switch position. But Raphael can help you with that ;)

Luznatural, 2022-10-31:
Here you have more data:
**Frsky X12S** (Only Hardware analog test as the raw is not working)
-100 0000
-60 0199
-20 0333
19 04CC
59 0666
99 07FF
**TX16S**
-100 0
-60 284-287 (last digit oscilating)
-20 597-600 (last digit oscilating)
19 907-910 (last digit oscilating)
59 1219-1221 (last digit oscilating)
99 1530-1532 (last digit oscilating)

Additionaly, the X9D+ Taranis may also have the 6 pos pots (optional mod) I have one... but it is not possible to get the raw information as it's not coded.

pfeerick, 2022-10-31:
> (Only Hardware analog test as the raw is not working)

What version of EdgeTX? The fact you're giving hex values sounds like it's an older version - need 2.6 or later to get the extra tabs and raw values. Or do you mean you're not getting an entry for the 6POS there? 

> Additionaly, the X9D+ Taranis may also have the 6 pos pots (optional mod) I have one... but it is not possible to get the raw information as it's not coded.

There is still a raw analogs page on ETX on B&W - use the page key once you're in the analogs menu, and you'll get the "Raw Analogs (5Hz)" option. If it went onto the S3 port I think offhand it should be analog 7, but you'll find out soon enough if it moves when at that screen.

Luznatural, 2022-10-31:
Thank you **pfeerick**
You were right. It was an old version of ETX.
I have updated de data for Frsky X12S above in the post with the ones of ETX 2.7.1. As soon as I come home, I will update the data for X9D+

rotorman, 2022-10-31:
@Luznatural is that TX16S Gen.1 or mkII that you added (just for background info)?

andytransonic, 2022-11-01:
Horus X12S, EdgeTX 2.8.0 nightly
505
819
1025
1364
2047
4091

Luznatural, 2022-11-01:
> 

Already updated the data on the post of three radios above

Luznatural, 2022-11-03:
When you do the improvement of the calibration screen, do not forget the task on https://github.com/EdgeTX/edgetx/issues/2108
to assure that the analog joystick of the X12S is also contemplated in the calibration. The owners of this radio (me included) will appreciate that. Frsky and OTX already sent us to oblivion. You (ETX developers) are now the only ones protecting us...
Very respectfully Thanks.

arthurson, 2022-11-05:
rm tx16s mk2
0, 257, 544, 833, 1098, 1385

Luznatural, 2022-11-18:
Any progress in this?
Thanks in advance

Luznatural, 2023-03-09:
I hope this issue continues moving forward, since last november no movement

rotorman, 2023-06-03:
Following commit implements default 6-pos calibration for TX16S and Boxer: https://github.com/EdgeTX/edgetx/commit/a9866f88e592d98bd79f91f49d2d14c715dddc42

neutor, 2023-12-15:
> Following commit implements default 6-pos calibration for TX16S and Boxer: [a9866f8](https://github.com/EdgeTX/edgetx/commit/a9866f88e592d98bd79f91f49d2d14c715dddc42)

Hi. Could you, please, add Pots calibration step? Currently I see no way to calibrate Pots. I fresh installed 2.9.2 on my Jumper T Pro and it went crazy, noises from speaker, unstable channel values jumping around ... The only difference I saw with previous radio.yml was calib: Pot1 (Pot2) section was missing on fresh install. 

And while Im here, my Jumper T Pro has a toggle switch instead of right pot. And it causes Direction: INV not work on it for some reason. S2 is a Pot in Radio / Hardware. Can we ha a separate target for ones with toggle switch?

pfeerick, 2023-12-15:
Pots (in fact all analog inputs) are calibrated during stick calibration.
Center the pots at the same time as the sticks, and move full range in the
second step, all g with the sticks.

With 2.10, there will be some capability for inverting pots/sliders and
some additional custom swirches also IIRC. Maybe this could be extended to
all swirches?

pfeerick, 2023-12-15:
Noise from speaker probably means you need to go to SYS -> Hardware and
enable the audio mute toggle.

neutor, 2024-01-12:
> Noise from speaker probably means you need to go to SYS -> Hardware and enable the audio mute toggle.

Thats true, it fixed the noise caused by EMI from transmitter board.

For the S2 being a pot instead of switch on Jumper T Pro, the following edit to radio.yml fixed the issue for me. I now can apply Mixers / Weight = -100 to effectively invert the output.

The wrong, default section in radio.yml:

```
POT2:
  mid: 0
  spanNeg: 0
  spanPos: 1009
```

The correct values that fix the issue:

```
 POT2:
    mid: 512
    spanNeg: 504
    spanPos: 504
```

After the fix, the S2 switch moves between 1000us and 2000us as it should.
https://intofpv.com/t-jumper-t-pro-s2-switch-shows-as-pot-in-edgetx

Should we have a separate target for Jumper T Pro with the switch instead of a pot in bottom right?

pfeerick, 2024-01-12:
May not need that in 2.10 with the flex inputs option. If it's not already something that is addressed by that now, it shouldn't be to hard to add something.

Tiefflieger68, 2024-03-18:
don't forget the mini stick at the foot of th RM MT12.
Isn't visible on cal screen too

pfeerick, 2024-03-18:
Yes, that is a pain as it is an option... i.e. you may have the switches or
the joystick fitted.