EdgeTX/edgetx #2363: 2.8.0-RC1 Tracker

pfeerick, 2022-09-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Stuff is not released on time

### Describe the solution you'd like

Release it on time already!

### Describe alternatives you've considered

Going back to OpenTX

raphaelcoeffic, 2022-09-26:
Is that 1. April already? 🤓

pfeerick, 2022-09-26:
Yup ;)

Stinky80, 2022-09-26:
hmm... bugTracker???

rlisch, 2022-09-26:
Downloaded latest Nightly to my touch screen upgraded Horus X12S. Now touch screen function is all over the place and not usable anymore. Returned to Saturday's release and all good.

raphaelcoeffic, 2022-09-26:
@rlisch you need to reset the config of the touch controller, see here: https://github.com/EdgeTX/edgetx/pull/2351

richardclli, 2022-09-27:
EL18 internal module settings missed the option to choose output type (PWM/PPM/SBUS/iBUS, etc.), this function is essential for EL18 to work properly with receivers.

richardclli, 2022-09-27:
Channel selector in NV14/EL18 screen, the left tab bar is quite small and too close to the left.  Not sure if any tuning can be made to this popup,

richardclli, 2022-09-27:
Should I please these findings in separate issue or attach to this one?

Eldenroot, 2022-09-27:
One issue per bug/request would be better I think--- _¨¨)

gagarinlg, 2022-09-27:
I agree, one issue per bug and just a reference here

jfrickmann, 2022-09-27:
https://www.youtube.com/watch?v=tbKQN6bVlAQ

pfeerick, 2022-09-28:
My X9D+ (*not* the 2019) is speaking with https://github.com/EdgeTX/edgetx/commit/6fd69f8053e807a1f70b7ede0d0c95c431e020a4, but it was with 2.8.0-rc1, so unless there is something else different with the X9E that may not be the cuplrit.

TheIsotopes, 2022-09-28:
what about #2025. Nothing has changed yet.

Gerold68, 2022-09-28:
> @Gerold68 this might be the cause of your issue: [6fd69f8](https://github.com/EdgeTX/edgetx/commit/6fd69f8053e807a1f70b7ede0d0c95c431e020a4). Since we don't use EEPROM anymore, the I2C bus is potentially not initialised, thus causing the volume setting to fail.

TNX

richardclli, 2022-09-29:
@pfeerick  I have added 2 issues #2411 and #2412, these 2 will affect the mass production of EL18 planned in Nov, please add these in v2.8.0, thanks.

pfeerick, 2022-09-29:
This is more an issue tracker for bugs reported for 2.8.0-RC1, so #2412 makes it onto this list. I would suggest it be targeted to be fixed for RC3, as think we have have enough things fixed to do RC2 shortly - particularly so we can get rid of the label EMs and Companion error. However, #2411 is on the 2.8.0 milestone.

yds, 2022-09-29:
flashed 6fd69f8 https://github.com/EdgeTX/edgetx/actions/runs/3141800150
volume works fine.. I have the same GF setting pot1 as volume on my X9E -- all good now. 🤘

TheIsotopes, 2022-09-29:
Ghost menu can no longer be operated via scroll key in latest RC, but it still worked with v2.7.1.

Eldenroot, 2022-09-29:
Would be nice to have another issue for tracking progress for LVGL rewrite of screens :)

raphaelcoeffic, 2022-10-01:
> Would be nice to have another issue for tracking progress for LVGL rewrite of screens :)

We have "Projects" for this, to which issues can be added. Not sure we have a (meta) issue that summarises which screen still need to converted.

closed 2022-10-02 by pfeerick