EdgeTX/edgetx #2365: Companion: Update Interface error on startup

pfeerick, 2022-09-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

After releasing v2.8.0-rc1, every time I run the 2.8.0-RC1 build of Companion, I get this:

[image]

regardless of what settings I have set - i.e. I can even have the update check set to "manual"

It doesn't appear if I run a prior `main` build of companion for win64. 😭 

### Expected Behavior

It not have the error!

### Steps To Reproduce

1. Install Companion for win64 from https://github.com/EdgeTX/edgetx/releases/tag/v2.8.0-rc1
2. Run Companion and be greeted with aforementioned error.

### Version

Other (Please specify below)

### Transmitter

Other (Please specify below)

elecpower, 2022-09-26:
On mobile so some guesses as to cause.
Error 301 is a URL redirect. The 1st step is to retrieve the release metadata and there has not been any redirect in the prior releases.
This maybe a temporary issue tied to the release changing status from draft or a cache not refreshed yet.

elecpower, 2022-09-26:
Ignore what I wrote above. The issue is triggered by compiler directive that has been actioned in the switch to release mode and the bug not picked up in testing as it worked like a dream.

PR #2378 (2022-09-28): fix(cpn): Start up interface error 301
Fixes #2365

Tested selfbuild and simulated release by setting environment variable EDGETX_VERSION_TAG
closed 2022-09-28 by pfeerick