EdgeTX/edgetx #2368: Spektrum Receiver Warning" Failsafe not set"

planemann2000, 2022-09-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On some (not all) of my models using Spektrum receiver, when loading I get a Failsafe not set up on the first screen on 2.8 with current 2.8.0 rc1, also had this warning in the nightly earlier. No failsafe mode button in the Model Setup page.

### Expected Behavior

No Failsafe setup with Spektrum Receivers 

### Steps To Reproduce

On some (not all) of my models using Spektrum receiver, when loading I get a Failsafe not set up on the first screen on 2.8 with current 2.8.0 rc1, also had this warning in the nightly earlier. No failsafe mode button in the Model Setup page.

### Version

Other (Please specify below)

### Transmitter

Other (Please specify below)

### Anything else?

Running Edge 2.8.0 rc1 In a Eachine TX16S Transmitter

planemann2000, 2022-09-26:
This issue will duplicate with Spektrum remote receivers on all my Flex Aura equipped planes.

pfeerick, 2022-09-27:
Can you zip your RADIO and MODELS folders from the radio SD card and attach them here? I'm not able to reproduce this, so need to know more about the configuration in order to see what is going on.

planemann2000, 2022-09-27:
Here are my radio and models folders.

Stan

pfeerick, 2022-09-27:
You won't be able to add them via email, GitHub doesn't accept any attachments... will need to drag and drop the file onto the text input area of the issue on GitHub.

planemann2000, 2022-09-27:
OK

Stan

planemann2000, 2022-09-27:
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/9650642/RADIO.zip)
[MODELS.zip](https://github.com/EdgeTX/edgetx/files/9650643/MODELS.zip)
This good?

pfeerick, 2022-09-27:
Yup, that worked. I'll have a look at that later today. Thanks :)

xxxntwv, 2022-09-27:
I have experienced the same issue using spektrum receivers.Any model I use that uses the dsm protocol causes the fail safe warning.

gagarinlg, 2022-09-27:
Nice, a Heisenbug

tonims1, 2022-09-27:
I have the same problem with a model v950 wltoys KN protocol.

xxxntwv, 2022-10-05:
Update on this issue,it still occurs on 2.8rc2.when switching from models with Frsky protocol to models using spektrum protocols is when it happens.switching between spektrum models it doesn't occur.

Mmcb23, 2022-10-12:
I don't get the warning, but there is no Failsafe setting in the model setup screen for DSM-based reseivers

planemann2000, 2022-10-12:
When the warning appears, it happens quickly after a switch out of position alert happens and the alert cleared, only on Spektrum Receivers.

Mmcb23, 2022-10-12:
I'm guessing I don't get the warning because I set up the models in Companion, so have Failsafe set. It is visible in the YML file. However, there is no option to set it in the radio.

raphaelcoeffic, 2022-10-12:
Gentlemen, as shown in #2527, it has nothing to do with Spektrum.

The specifics are the following:
- internal MPM
- previous model should have failsafe, but it's not set
- switching to a new model that use the internal MPM, and does not have failsafe settings

What happens is basically that the status info from the previous model/protocol are used to assess the failsafe settings of the new model, which is obviously wrong.

pfeerick, 2022-10-12:
> I don't get the warning, but there is no Failsafe setting in the model setup screen for DSM-based receivers. Same behaviour in Simulator

Also, just FYI, the bug there is on Companions side, because Spektrum doesn't have a transmitter configurable failsafe, it's receiver side only (forward programming receivers are a different matter entirely). However, Companion/simulator can't query the MPM to find out what protocols support failsafe settings, etc, hence why the option is still shown.

planemann2000, 2022-10-12:
I have tried to get the failsafe warning for the past 2 days with no success on either transmitter I have (eachine tx16s nor the radiomaster tx16s, both running 2.8.0 rc2) I haven't used Companion yet. I'm stumped, but I have no problem to work around this issue.

pfeerick, 2022-10-12:
Raphael caught the bug shortly after asking how to repo after we narrowed it down to the internal MPM. Hence the linked PR #https://github.com/EdgeTX/edgetx/pull/2527 which will be shortly merged, and in the next nightly.

closed 2022-10-12 by pfeerick

xxxntwv, 2022-10-15:
This issue appears to be fixed in RC3

PR #2527 (2022-10-12): fix: clear internal module FIFO after DMA init
Otherwise, NDTR will be 0 instead of [buffer size], and thus the read index will not be set to the correct value.

Fixes #2368