EdgeTX/edgetx #2372: Font space width differs between <2.8 and 2.8

gagarinlg, 2022-09-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The space sizes are no dynamic, depending on the font size, this breaks some Lua scripts with pixel excat positioning of things

### Expected Behavior

?

### Steps To Reproduce

Start a Lua script

### Version

Nightly (Please give date/commit below)

### Transmitter

Other (Please specify below)

pfeerick, 2022-09-27:
What say you @JimB40 @jfrickmann @stronnag @yaapu @CapnBry ?

Fixed width spaces or font size proportional spaces?

gagarinlg, 2022-09-27:
No fixed width, but closer to the 4 pixel. Fixed width wouldn't work without manually/scriptually (is this a word?) editing the generated lvgl font files, which I refuse to do.

CapnBry, 2022-09-27:
Proportionalllllllllllllll if someone is using spacebar spacebar spacebar to line up their elements, they're their own problem :-D

gagarinlg, 2022-09-27:
I totally agree, but I think some just do not have enough room for it.
How about adding the different space widths to the fonts, that's costs only a few bytes of flash for the descriptor and map all of those codepoints to the standard space for B&W radios?
Then the Lua people (sounds like a band to me) can decide which space they want. Except that that totally breaks compatibility with older versions.

raphaelcoeffic, 2022-09-28:
Guys, do we really want to support broken stuff for ever? ATM, we are already providing backward compat for broken font baselines. Do we really now want to have support for broken spaces?

gagarinlg, 2022-09-28:
> Guys, do we really want to support broken stuff for ever? ATM, we are already providing backward compat for broken font baselines. Do we really now want to have support for broken spaces?

I would prefer having the 0x20 space as it is default for that font and add 0x2009 (THIN SPACE) and 0x200A (HAIR SPACE) to the fonts, so that it is possible to select smaller spaces, when needed.
The other space variants are also proportional, but smaller than the normal space.
Do we all, including the people that have to adapt their scripts, agree to do it that way? OpenTX is mostly dead anyways (now I am going to get hit) and we have to start cleaning things up. There is not enough man power and masochism available to maintain compatibility to broken stuff for ever.

pfeerick, 2022-09-28:
For ever - no... that's what 3.0 is expected to break a lot of... that's the time to break all the old APIs ;) I have no problem with the proposed suggestion, but I'm not a script author ;)

JimB40, 2022-09-28:
@gagarinlg I have yesterday nightly with your PR 
So to confirm. Both on 2.7 and 2.8 have now 4px spaces?
While 2.7 looks like it has 2.8 nighly seems to have proportional

[image]

[image]

gagarinlg, 2022-09-28:
I did not make any change to the space width yet, I am waiting for the discussion to finish.

JimB40, 2022-09-28:
Thing is as you can see on screenshot that if space settings are the same (4px) it seems LVGL renders space properly.
Moreover I've checked 4-5 scripts (not INAV) and all of them look okay. 
To my eye 2.8 (LVGL) is  1-2% mode compact (kerning). But with different render enginge I doubt we will pinpoint extact same look. 

Left 2.7.1 - Right 2.8 nightly

[image]
[image]
[image]

JimB40, 2022-09-28:
Yes here is a proof it's slighlty more compact. Pay attention to multi line text boxes on the right 
[image]

gagarinlg, 2022-09-28:
The kerning difference is probably, that before 2.8 no real kerning was done.

gagarinlg, 2022-09-28:
> Yes here is a proof it's slighlty more compact. Pay attention to multi line text boxes on the right (SMLSIZE used) [image]

There two things come together. The space width for SML is slightly smaller with 2.8 (3 instead of 4) and the kerning differences.
I do not think we will get it to render pixel exact identical on 2.8 as it was before

JimB40, 2022-09-28:
And I can see now lineHeight is rendered bigger on 2.8. At least for SMLSIZE font size (13)
You can see that on screenshot and because it's rendered bigger last line with 'wrong target' disappeared.
But this will affect only multiline fields.
If .line_height is equal to 13 in both 2.7 & 2.8 it's about LVGL render engine 

[image]

gagarinlg, 2022-09-28:
This can probably be set somewhere for the multi line text field

JimB40, 2022-09-28:
For me is not a problem, using multi-line twice.
But let others speak. Just run scripts you have and look if things look good.

gagarinlg, 2022-09-28:
> I checked my scripts and text placement is ok, except for the known issue about the v-align and left offset
> 
> [image] [image]

when you use a nightly the vertical text alignment and the 1 pixel offset should be fixed

jfrickmann, 2022-09-29:
> Guys, do we really want to support broken stuff for ever? ATM, we are already providing backward compat for broken font baselines. Do we really now want to have support for broken spaces?

I would say that it should be fixed once and for all, and then we just have to suck it up and update our Lua scripts. But this means that the fix should be tested carefully, so we do not see a number of changes and have to change the Lua scripts over and over again. And if we have to update the Lua scripts, then let's get rid of that offset business in Lua as well!

As a general philosophy, I think that we should keep it backwards compatible if feasible. But sometimes this is not possible, and then we just have to update our scrips - that's the price of progress!

Jesper

stronnag, 2022-09-29:
Hard choices. 

If one maintains a complex script that works on both OpenTX and EdgeTX and the majority of users are on OpenTX, then there will be a difficult decision to be made if maintaining backwards compatibility is not easy.

gagarinlg, 2022-10-02:
As @JimB40 is happy the way it is now, when none of you is requesting changes, I will leave it as it is. If someone really needs a change of the space sizes, I will add the smaller space characters and you can use those.

If no one complains until tomorrow evening (my timezone, CEST/GMT+2), I will close the ticket

gagarinlg, 2022-10-03:
Time is up. This issue will be closed and script writers have to adapt to the LVGL font renderer space sizes.

closed 2022-10-03 by gagarinlg