EdgeTX/edgetx #2373: Model Necromancy

gagarinlg, 2022-09-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When upgrading from an earlier version, which has unreferencedbmodels on the SD card, those models are imported again.
This may cause problems for people who delete many models by creating duplicates. 

### Expected Behavior

That is to be discussed.
Leave it as is or automatically delete all unreferenced model files and then models.yml

### Steps To Reproduce

Upgrade radio with unreferenced model files

### Version

Nightly (Please give date/commit below)

### Transmitter

Other (Please specify below)

pfeerick, 2022-09-27:
I would suggest in the first instance leaving it as it is (i.e. the initial import will just import all, and force you to deal with the issue properly to start with). Especially as everything is going to get chucked into the "Unlabled" label anyway. 

I'm in two minds as to the "everyday" behaviour... as I like the idea of being able to "undelete" a model. But how to action this? Deleting a model doesn't delete a model, but labels it deleted, and if you delete it then it goes bye bye? On B&W, Companion will culls all `model##.yml` files, since they will otherwise "come back" when you do a read/write cycle with Companion, since it didn't use the index file but blindly reads the files in order. 

Perhaps we finally fix/restore the "Backup/Restore" functionality, so that deleting moves a model to a Backup folder (and out of the models folder so it won't be rescanned), and you can then "Restore" or "Delete" via the SD browser on radio?

jmxp69, 2022-09-27:
I don't think you lose any functionality by actually deleting the file. Regardless of prior behaviors, I get the sense if someone deleted the model they believed it actually gone forever regardless of what existed or didn't exist on the card itself. 

I think it makes sense to create a deleted bin somewhere on the card with an option to un-delete. I do not think that option should exist by moving files around on the card manually as there exists potential for the user to overwrite or duplicate filenames creating a potential mess.

Obviously advanced users will be able to figure out how to relocate/rename files manually, but I don't think it should be the standard recommendation.

rburrow87, 2022-09-27:
> I would suggest in the first instance leaving it as it is (i.e. the initial import will just import all, and force you to deal with the issue properly to start with). Especially as everything is going to get chucked into the "Unlabled" label anyway.

An issue that the previous version(s) of EdgeTX caused unbeknownst to the user 😉 When it first happened to me, I had duplicate models that had the same name and image because I had done something like updated some layouts/logic in one model and duplicated it to replace another model file so that they matched without having to re-do the work. But maybe that's not something normal people do?

Of course, I had companion backups, so it was no big deal to nuke the models folder on the TX, open my backup file in Companion 2.8, and restore it to the TX to sort it out. I see a certain someone already made a video about it 👌

jmxp69, 2022-09-27:
>  When it first happened to me, I had duplicate models that had the same name and image because I had done something like updated some layouts/logic in one model and duplicated it to replace another model file so that they matched without having to re-do the work. 
> 

I definitely think this is a problem. If a user makes changes to a file and winds up with a dupe, how do we ensure they keep the correct file. I believe the answer to this lies in models.yml. Scan for model???.yml names, move anything not  referenced there to a holding directory and keep only referenced models in models.yml in \MODELS.

gagarinlg, 2022-09-27:
I dug a little into the code. To "fix" this we would need to

- re-add storage/modelslist.*, storage/yaml/modelslist.*
  - remove all add/move/rename stuff from there
- at every let the modelslist code run
  - check if there is a modelslist
  - move the modelslist and all model files that are not mentions in the modelslist to another directory

all of this while still be able to read multiple variants of the modelslist and bin and yaml model files.

pfeerick, 2022-09-28:
> An issue that the previous version(s) of EdgeTX caused unbeknownst to the user 😉 

Have to go back much, much, much further than that... this crazy system has been like that for a long time, and used to confuse me in the early days - was part of the reason you had a whole lot of bin files when you only had a handful of models, so it was a game of Russian roulette trying to figure out which was your actual model file. They should have been moved into a `deleted` directory if they wanted to adopt the "protect the user from their own actions" model as is done in most operating systems. 

> I believe the answer to this lies in models.yml. Scan for model???.yml names, move anything not referenced there to a holding directory and keep only referenced models in models.yml in \MODELS.

Yup, that and `models.txt` in both `/MODELS` and `/RADIO`, and moving anything not referenced to an `UNUSED` subfolder is probably the only way to deal with this. Maybe also a prompt if it is triggered? 

ie.
```
Unreferenced models found!
Check your radios MODELS/UNUSED
folder to restore or delete them. 
```

or words to that effect. With suitable documentation/video to support it. ;)

dlktdr, 2022-09-28:
> I'm in two minds as to the "everyday" behaviour... as I like the idea of being able to "undelete" a model. But how to action this?

As of right now when you delete a model it moves it into a subfolder "DELETED" so if you delete one you have a chance to recover it incase of accident.

What @gagarinlg says seems like a good solution. I can work on this. Will re-add the (simplified) models list code, Move anything not referenced into another folder "MODELS/UNUSED".

What about companion, since it just opens all modelXX.yml files now ignoring the models.yml file too. Same idea if done there?

Was trying not to keep unnecessary extra code, didn't think about the extras floating around in there 😉

gagarinlg, 2022-09-28:
Companion does not need it.
Before the companion has a chance to access the files after an update, the radio should have finished the conversion.

pfeerick, 2022-09-28:
> As of right now when you delete a model it moves it into a subfolder "DELETED" so if you delete one you have a chance to recover it incase of accident.

Yeah, I saw that folder afterwards... so I guess a later feature can be to allow "Import/Restore" of a model via the SD browser so the firmware manages it being put back into place so there are no filename collisions, etc. Probably need to make the file browser preview at least show the model name as well as `model99.yml`, isn't very... helpful. 🙃 

(Unfortunately) what happens to the radio needs to happen on companion also, so yup, same deal there. I know what you mean about unnecessary code... unfortunately this seems to be necessary as John duplicates stuff a lot! 😆 (I'm just as guilty... the scan on my radios resurrected a fair amount of dead wood also) .

pfeerick, 2022-09-28:
> Companion does not need it.

I'm not so sure - since there is more than just exclusively working with the radio SD data - i.e. what happens when you load the saved pre 2.8 `.etx` into Companion 2.8 and want to write that to the radio? Neil @elecpower would be the best one to comment on Companion, but generally, Companion should always have the same behaviour/capabilities as the radio, as it is expected to behave the same way. If it were exclusively let radio firmware do something, read from the radio, write back to radio, that would be fine. But there are all sorts of Simulator and Companion usage cases I hadn't even thought of until they were mentioned.

gagarinlg, 2022-09-28:
> what happens when you load the saved pre 2.8 `.etx` into Companion 2.8 and want to write that to the radio?

The radio will sort it out, when it can access the SD card again. I think there is no harm done in that situation.

dlktdr, 2022-09-28:
I can see the situation where you would open and old .etx file before opening on the radio, it would import every modelxx.yml. If I then add a new one and save, even if the original models.yml file was copied this new model would get moved to /models/unused on boot, as it currently stands. I can't say for sure that models.yml even stays anymore. Tests for tomorrow, late here.

I can work on companion to do the same right after the FW change, jic.. I know I go straight to the radio (and apparently keep my dir clean 😆 ) but for minimal complaints will do it here too. 

Hopefully won't hold this release back too much 😬..

jmxp69, 2022-09-28:
> ```
> 
> or words to that effect. With suitable documentation/video to support it. ;)

On it as soon as the fix is ready.

dlktdr, 2022-09-29:
On the companion front, 

Looking into this in more detail, when you do a radio backup, only the models that were listed in models.yml would have been saved in .etx format in the first place. @gagarinlg is correct. I thought it did a copy of the folder, I was wrong. So it appears nothing needs to be done on the companion front.

pfeerick, 2022-09-30:
Yes, it looks like Companion from even 2.5 and 2.6 days would only look at the models in `models.txt` and `models.yml`, and only those models were saved into the model zip (otx/etx file) (and sequentially numbered). So it would only be a otx/etx someone has tinkered with, and their own their own then! 😆

closed 2022-10-01 by pfeerick

dlktdr, 2022-10-01:
I just thought of something that might cause another bit of grief.. the models.yml file is copied into /unused now.. it would be pretty easy for a user to just copy this back into /models by accident / no knowing, etc.

Should it be moved and renamed to /models/unused/models.yml.old instead, so if it's copied back in it wouldn't be re-parsed potentially after user has made changes in the meantime?

pfeerick, 2022-10-01:
Probably wouldn't hurt. It's one of the reasons I would prefer model restoration be done through the radio SD UI... i.e. browse to models/UNUSED, the preview will tell you the name of the model as you scroll through the files, and you could then long press and "Import Model" or some such... thus removing the desire to wildly copy and paste stuff around. But that is a future UI enhancement.

dlktdr, 2022-10-01:
Yeah that sounds like a good idea, In the future. 

For this next RC candidate I'll change the extension. Pretty easy one.. PR #2430