EdgeTX/edgetx #2376: In the flight modes menu, can't tell which trim is being edited

RC-SOAR, 2022-09-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

TX16S, EdgeTX 2.7.1

The Flight Modes menu has six fields for editing the properties of the trims. However the fields lack a label, so it's impossible to know with any certainly which field goes with which trim (my guess is that the order is AETR, T5, T6...  but it's only a guess...).

Maybe have a submenu for the trims which would give more room for explanatory text. Or else a label at the top which displays the name of the trim with the focus.

I'm calling this a bug rather than a feature request, because the feature is already there, but is missing an essential element.

### Expected Behavior

See above

### Steps To Reproduce

See above

### Version

2.7.1

### Transmitter

Radiomaster TX16S

pfeerick, 2022-09-27:
Have a look at 2.8-rc1 and see what you think ;)

[image]
became
[image]

RC-SOAR, 2022-09-27:
Looks good!

pfeerick, 2022-09-27:
Fantastic! Any suggestions you have regarding the UI would be appreciated as we move onto 2.9 and finishing the LVGL migration.

closed 2022-09-27 by pfeerick