EdgeTX/edgetx #2379: cpn: simulator won't let me delete previously duplicated theme

mha1, 2022-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Can't delete duplicated theme in cpn simulator. Will crash companion sometimes after switching back and forth between themes if attempt to delete was made previously.

Note: haven't checked radio yet

### Expected Behavior

being able to delete themes.

### Steps To Reproduce

1. go to radio settings and select THEMES
2. select THEME, e.g. Greenfields
3. duplicate theme with name n, author a, description tbd and save
4. make attempt to delete theme n

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

### Anything else?

self build W64 Companion on latest commit (738615549

pfeerick, 2022-09-28:
I can reproduce this on Companion simulator, TX16 actual hardware seems to be working fine - admittedly different build (from #2371) but that *shouldn't* matter. Surprisingly `simu` also worked. So I think this _may_ have to do with the fancy path handling that happens when running Simulator through Companion.

I did manage to get this error out of it (where `E:/Documents/EdgeTX/RadioMaster TX16s/SD Card 2.7 Tester/THEMES` was the correct path) ... don't know why a permission error held it up but maybe that has some bearing on this. 

```
604.21s: INDEV_STATE_RELEASED
606.47s: f_rename(E:/Documents/EdgeTX/RadioMaster TX16s/SD Card 2.7 Tester/THEMES/a/theme.yml, E:/Documents/EdgeTX/RadioMaster TX16s/SD Card 2.7 Tester/THEMES/a/theme.yml.deleted) = error 13 (Permission denied)
606.47s: opening directory: /THEMES
606.47s: f_opendir(E:/Documents/EdgeTX/RadioMaster TX16s/SD Card 2.7 Tester/THEMES) = OK
606.47s: scanForThemes: open successful
```

mha1, 2022-09-28:
Just a wild guess: was the file closed before the attempt to rename it?

mha1, 2022-12-22:
still open in 2.8

pfeerick, 2024-03-04:
I think #4148 confirms this is present in firmware also, so isn't simulator specific.

philmoz, 2024-03-05:
> I think #4148 confirms this is present in firmware also, so isn't simulator specific - or is otherwise a very closely related bug.

I think these are different issues, this one is about crashing when deleting a duplicate theme (not reproducible).

4148 is about not being able to create a new theme with the same name as a deleted theme - this happens because the folder still exists when a theme is deleted so the duplicate action fails.

pfeerick, 2024-03-05:
Yeah, on second look similar but not the same. I was able to reproduce it (on Windows). I cannot reproduce on Linux, which further reinforces my earlier suspicion about it being related to path handling in Companion/Simulator.

mha1, 2024-10-13:
seems to be fixed

closed 2024-10-13 by mha1