EdgeTX/edgetx #2383: Companion treating X12S ACCESS ISRM like XJT

rburrow87, 2022-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I just realized I never submitted this as a bug and was probably forgotten in a PR comment from a while back https://github.com/EdgeTX/edgetx/pull/2019#issuecomment-1207302870

>Okay now that you made me look at my old post, maybe it's caused by something else 😛 Almost seems like it's treating the X12S like the internal module is set to XJT instead of ISRM even though it's set to ISRM.
>
>This is what the X12S ISRM radio writes:
>```
>moduleData: 
>   0:
>      type: TYPE_ISRM_PXX2
>      subType: ACCESS
>      channelsStart: 0
>      channelsCount: 16
>      failsafeMode: NOPULSES
>      mod: 
>         pxx2: 
>            receivers: 0
>            racingMode: 0
>            receiverName: 
>               0:
>                  val: "SR6"
>```
>
>This is how Companion 2.8 reads it in (notice it offers the `Antenna` option as if it's treating it like the internal module is set to `XJT` and not `ISRM` (and I made sure it's set to `ISRM`)):
>
>[image]
>
>So this is how Companion 2.8 writes it back out:
>```
>moduleData:
>  0:
>    type: TYPE_ISRM_PXX2
>    subType: ACCESS
>    channelsStart: 0
>    channelsCount: 16
>    failsafeMode: NOPULSES
>    mod:
>      pxx2:
>        receivers: 0
>        racingMode: 0
>```
>
>But Companion 2.8 still reads in and writes out fine with my X9D+2019:
>
>[image]

### Expected Behavior

for it to work

### Steps To Reproduce

do the steps

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12

pfeerick, 2022-09-30:
Need to consider https://github.com/EdgeTX/edgetx/pull/1765 and associated PR for that latter point I think.

elecpower, 2022-09-30:
Looks like best for me to wait until 1765 is merged and see what Companion tweaks are necessary to match the radios

PR #3050 (2023-02-27): fix(cpn): hardware and internal module antenna parameters
Fixes #2383

Summary of changes:
- align Companion with #1765
- hardware
-- move antenna to within Internal RF section
-- code housekeeping
- model setup
-- Internal module fix when to display antenna parameter
-- fix antenna dropdown list sequence
- boards fix external antenna

Tested on a few libsims
closed 2023-02-27 by pfeerick