EdgeTX/edgetx #2384: 2.8.0-rc1 Logical switches monitor becomes unresponsive

RC-SOAR, 2022-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a setup with 36 logical switches.  This is the sequence:

1. From the Logical Switches menu, I press the MDL button. This opens the Channels monitor.
2. Touch the icon to open the Logical Switches monitor. 

At first, the Logical switches monitor shows a correct representation of the logical switch states.

Now... I play with the physical switches which alters the logical switch states.  _After a few switch operations, the monitor becomes unresponsive (EdgeTX stops updating the monitor display)._  

I can tell that the logical switches are working correctly because they trigger special functions.
I am able to exit the monitor screen as normal.
The problem is repeatable without rebooting.

TX16S / 2.8.0-RC1

### Expected Behavior

The monitor should display the current state of the logical switches.

### Steps To Reproduce

See above

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

YAML file:

[model4.yml.txt](https://github.com/EdgeTX/edgetx/files/9659980/model4.yml.txt)

JimB40, 2022-09-28:
On my TX16S Logical Switch monitor doesn't work at all. Nothing is highlighted.

banyaszg, 2022-09-28:
On the Logical switch monitor it is not highlighted, only the text color changed. (I think from primary to secondary.) It is slightly noticable, and I assume the background should change.
I tested with two simple logical switch connected to a physical switch value, and the first change was displayed, but no refresh after that. I'll check the code when I have some free time.

RC-SOAR, 2022-09-28:
@banyaszg Yes, active switches need to be better highlighted.

banyaszg, 2022-09-28:
It seems the code contains an attempt to color the background, just the used mechanism is not implemented in the libopenui. Also it is strange that the implementation uses a bunch of TextButtons instead of a ButtonMatrix. I'm going to work on this tonight.

pfeerick, 2022-09-29:
Not all the code has been ported over to LVGL constructs, so this probably won't be the only such thing... but is probably the case I've seen of late ;)

banyaszg, 2022-09-29:
@pfeerick So it seems to me in the old version the font seelection of the textFlags was used every time when the drawText() was called for the device context. In the LVGL version it is used only at construction time to set the font to the main part. And the setTextFlags() sets only colors. But because of the old behaviour it is used here, at the logical switch monitor and 3 other places to set the font.
Which would be the better way, implementing the font selection into the setTextFlags(), or rewriting these occurrences?

pfeerick, 2022-09-29:
First I'm going to premise this with the fact that @raphaelcoeffic and @gagarinlg are the LVGL whispers here as I haven't been involved too much in the porting, nor do I understand a lot of it still... Looking at the effected code, it looks like it's only this screen, and the LS and SF header indicators that want to "actively" change the font... the rest of the instances are all static? So I think a re-write is in order, given the use of `LV_STATE_USER_1` type code in the more recently written `model_ouputs.cpp`. btw, hint, that means the similar `LV_STATE_USER_1` change you asked about in the PR will probably be fine ;)

banyaszg, 2022-09-29:
I created a separate issue #2414 about this textFlags thing.

PR #2410 (2022-10-02): fix(color): Logical switches monitor fix
Fixes #2384

Summary of changes:
This patch uses styles for active and non active buttons. The style defines colors and fonts for each logical switch state.
closed 2022-10-02 by pfeerick