EdgeTX/edgetx #2385: Incorrect wrap behaviour in Mixers menu when using scrollwheel

RC-SOAR, 2022-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In the mixers menu, when using the scrollwheel, the mixer list wraps to be beginning when the end is reached. The 'Show mixer monitors' option, and '+' lines wrap as well. 

### Expected Behavior

Wrap behavour with scrollwheel should be consistent with touch (which works correctly).

### Steps To Reproduce

Open the Mixers menu and - using the scrollwheel - scroll past the end of the mixer list.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

TX16S / 2.8.0 rc-1

raphaelcoeffic, 2022-09-28:
> So the problem really is that the 'Show mixers monitors' and '+' options scroll and wrap as well. I think these would be better as static elements.

What do you mean with "static elements"? If you mean to have the mixer channels/lines in a separate scroll-box, and these 2 always on screen, I'm afraid this going to cost quite some space, thus leaving much less for actually displaying the mixer channels/lines (which some people have been complaining about).

RC-SOAR, 2022-09-28:
By 'static', I mean it shouldn't scroll. I appreciate that an extra row would not be practical, but there appearst to be space on the title bar for extra info. This could also be used for the mixer monitor display:

[image]

gagarinlg, 2022-09-28:
But then you only have one mixer monitor nad not monitors for every mixer that fits onto the screen

The +-Button in the title bar to add a new channel might work well

RC-SOAR, 2022-09-28:
> But then you only have one mixer monitor nad not monitors for every mixer that fits onto the screen

If want to see all the mixer values, a single press on MDL takes you to the Outputs,

It's not a new idea, OpenTX displays the value of just one channel:

[image]