EdgeTX/edgetx #2386: Deleting a label causes freeze / EM

rburrow87, 2022-09-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When deleting a label, the progress pop up comes up but never moves. If you try to exit out, controls are unresponsive. From there, depending on the button/screen mashing, it'll either go into EM or you have to pull the battery.

### Expected Behavior

Should just delete.

### Steps To Reproduce

1. Create a label
2. Delete the label
3. ???
4. EM

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X12, Radiomaster TX16S

### Anything else?

Clearer with video:
https://www.youtube.com/watch?v=U0vE26PajUo

pfeerick, 2022-09-28:
Thank you, thank you... I'm not going nuts then... As I was sure I had managed to do this before, but couldn't remember how to repo the problem. In my case it may have been deleting an *empty* label (or perhaps an active one?), as if I create, power cycle, and then try to delete that label, basically have the power cycle the radio to prevent an EM.

dlktdr, 2022-09-28:
I'm on it... Thanks for the video..

PR #2392 (2022-09-28): fix(color): Deleting empty label caused EM/Hang
Fixes #2386

Looks like I introduced this with removing YAML characters to prevent unwanted characters in the Label. When deleting labels, it does a rename of all the models labels to "". There was a check that was preventing this action because size was zero..

Ooops.

**Edit:** Went into autopilot on the last PR.. Should be put into 2.8
closed 2022-09-28 by pfeerick