EdgeTX/edgetx #2393: Please add flysky fs-i6s

timelord666, 2022-09-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have the fs-i6s and i want to increase it's range, so i want to use elrs. But there's no firmware for fs-i6s and i don't know why, it's great tx, well with minuses but, still good.

### Describe the solution you'd like

just common firmware that can work with elrs

raphaelcoeffic, 2022-10-03:
The reason why is that the MCU on this radio is way too small. I think Someone ported an old release of OpenTx to it, with lots of features stripped down.

closed 2022-10-03 by raphaelcoeffic

pfeerick, 2022-10-04:
I thought that also, but I think it's for a different flysky tx 😢

MightyPork, 2024-03-13:
I have a i6s on order so while I can't promise anything I will at least open it and look what the specs really are, if the MCU part number is not concealed. There's very little information about it online - but I think it can't be that bad given they support a touch screen and a matrix graphic display?

pfeerick, 2024-03-13:
https://github.com/OpenI6X/opentx is the supported opentx firmware fork for the Flysky FS-i6X. There has been some discussion about also supporting the FS-i6S, but the issues there are

1. there is no going back to the stock firmware after installing anything else - the factory firmware is read protected, and the flysky update process appears to reply on some special boot mode so you can't even use the firmware updater to get back to flysky firmware.
2. black and white UI on opentx was never designed with a touch interface in mind, and there is only the power button and two buttons on the back of the radio as hard keys. So in addition to adding support for the touch screen (not likely to be any major issue) we need to figure out how to use the touch screen with the current UI

Come join the [Discord server for the OpenI6X project](https://discord.gg/hmK5R6sR), and have a look at the [discussion from January in the `#development` channel](https://discord.com/channels/973289741862727741/973290217266118677/1198244514410803321). The firmware for the I6X actually works on the I6S, but obviously with no buttons for navigation, can't do anything with the radio.

MightyPork, 2024-03-13:
Thanks for the link, that looks exciting. I'm only just getting into RC, so bricking my first controller for experimentation with firmware changes is not really on my roadplan haha 

Though I have background with microcontrollers and FreeRTOS so maybe I will be able to help somehow, down the line