EdgeTX/edgetx #2398: mixer lines cannot be copied or moved in the UI of the radio

ruedli, 2022-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When in model settings, the mixer tab I cannot copy or move mixer lines to different positions.

The popup menu show "edit" / "copy" "move" / "insert after" / "insert fefore" / "clear" etc. : I select copy.

After this I cannot insert or paste this copied line . Insert after, inserts a default line, not the copied one.

### Expected Behavior

After having selected a line with "copy" or "move", I expect the next select to offer "paste"/ "paste after" / "paste before" options.

"paste" should replace the selected line, "paste before" and "paste after" should insert a mixer line at the appropriate place.

If the selected line was the result of a "move", the original mixer line should be deleted.

### Steps To Reproduce

1. select "model settings"
2. select the "mixers" tab
3. select a mixer line and at the popup select "copy" or "move".
4. there is no way to use this copied line....

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

version v2.8.0-rc1

I believe the explanation is clear, without screenshots, it affects the radio user interface, not the companion. If fotos are needed, let me know.

I used the provided firmware for my radio, as well as built it myself using the release tag v2.8.0-rc1, with the -DLUA_MIXER=YES
Build environment used:

https://gitpod.io/#https://github.com/EdgeTX/edgetx/releases/tag/v2.8.0-rc1
cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DHELI=OFF -DCMAKE_BUILD_TYPE=Release ../

both (provided firmware / my own build) had the same behavior

pfeerick, 2022-09-29:
Pretty sure it used to say "Paste" before after doing a "Move". Same goes for "Copy". Reproducible on TX16S

eshifri, 2022-09-30:
There is the same problem with the Inputs screen.

PR #2423 (2022-09-30): Fix Copy/Move/Paste mixer line
Fix functionality of moving and pasting mixes.

Fixes #2398 

Summary of changes: use pasteMix() for mixes rather than pasteInput(). (Kind of make sense.)
Use the relevant class member instead of global variable to detect the fact that a line was saved for Paste/Move.
closed 2022-09-30 by pfeerick