EdgeTX/edgetx #2400: Improvements to Mixers menu UI

RC-SOAR, 2022-09-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Channels without mixers are not displayed. This leads to two issues:

1. There are two entirely different methods of input for new mixers, depending on whether the channel is displayed:

- If the channel is displayed, then it's {long touch}, then {Insert before/after}. 
- If a channel is not displayed, then scroll down to the bottom and touch {+}, then choose a channel.

This is confusing.

2. It means you can't see the 'holes' in the channel list. For the complete picture, it should be possible to see _all_ channels.

### Describe the solution you'd like

I would like to revert to the OpenTX display format, where all channels are displayed, whether used or not. Then only {insert before/after} would be needed.  

Also, I think it would be more useful to display channels in the format CHnn:[name] - channel numbers are useful to see, since they can be the source of a new mix.

Ultimately, there should be no distinction between channels with and without mixes, since internally they all contribute to the outputs.

### Describe alternatives you've considered

Another option would be to just have just a single method of adding mixers, via an enhanced '+' option. 

Pressing on '+' would open a submenu displaying a list of all locations to place the new mix. 

However this would IMO would only work well if the '+' option was permanently displayed (currently, to access the '+' option, you have to scroll down to the bottom of the channel list).

### Additional context

TX16S / 2.8.0 RC-1

gagarinlg, 2022-09-28:
> Also, I think it would be more useful to display channels in the format CHnn:[name] - channel numbers are useful to see, since they can be the source of a new mix.

I agree, the number should also be shown, but real estate is limited, so nn:[name] should be enough. 

> Ultimately, there should be no distinction between channels with and without mixes, since internally they all contribute to the outputs.

Without a mixer there is no data source for the output, so it just transmits its center value all the time. To keep the UI clean I think it is totally reasonable to not display channels with no mixer.

> However this would IMO would only work well if the '+' option was permanently displayed (currently, to access the '+' option, you have to scroll down to the bottom of the channel list).

This is also a reals estate problem. When the + Button is shown all the time, less mixers can be displayed at the same time.

RC-SOAR, 2022-09-28:
> This is also a reals estate problem. When the + Button is shown all the time, less mixers can be displayed at the same time.

Looking at the 'MIxer' title bar, there appears to be quite a bit of space which is not currently used. Could this not be put to use?

[image]

pfeerick, 2022-09-28:
Also need to juggle making stuff big enough for touch, and non-touch users.

gagarinlg, 2023-05-05:
@RC-SOAR did you check the current nightlies or 2.8.3? It looks much better now.

RC-SOAR, 2023-05-05:
Not yet, but look forward to trying this next week.