EdgeTX/edgetx #2403: 2.8.0-RC1 - BREAKSPACE ignored in "Aggiornamento riuscito" after flashing MPM firmware (ITA)

robustini, 2022-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I don't think it is a problem even in the languages with missing BACKSPACE, but it is certainly due to the need to have a BACKSPACE that separates the two words in Italian because it is necessary when instead you write others firmware from the tx.
From the it.h (my traduction), line 830:

#define TR_FIRMWARE_UPDATE_SUCCESS     "Aggiornamento" BREAKSPACE "riuscito"

### Expected Behavior

The backspace should work, the two words should be one above and the other below.

### Steps To Reproduce

1. Try to flash MPM firmware with the italian translation

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

[image]

robustini, 2022-09-29:
Same problem in other contexts where BREAKSPACE is used, verified now.

raphaelcoeffic, 2022-10-01:
This is using some legacy feature that is only available on B&W screen radios:
```
#define BREAKSPACE           "\036"
```

On color radios, the label placement/wrapping is handled by LVGL depending on the space given to the label, so it is basically obsolete and should be redefined to a simple space character (if this is what is intended).

robustini, 2022-10-01:
@raphaelcoeffic so for example what do you think about changing this::

#define TR_NO_SOUNDS_ON_SD "No sounds" BREAKSPACE "on SD"

In this:

#define TR_NO_SOUNDS_ON_SD "No sounds" BREAKSPACE "on SD" , "No sounds on SD"

Correct or obsolete?

raphaelcoeffic, 2022-10-01:
Probably more along these lines (in a more generic header file, common to all translations):
```c++
#if !defined(COLORLCD)
  #define BREAKSPACE           "\036"
#else
  #define BREAKSPACE           " "
#endif
```

robustini, 2022-10-01:
> Probably more along these lines (in a more generic header file, common to all translations):
> 
> ```c++
> #if !defined(COLORLCD)
>   #define BREAKSPACE           "\036"
> #else
>   #define BREAKSPACE           " "
> #endif
> ```

So do you suggest doing this in each translation file, or directly in the code elsewhere?

robustini, 2022-10-01:
I think it is correct to leave the current syntax but by reversing the condition, I don't understand why it is now the other way around.
So it would become:

#if defined(COLORLCD)
  #define INDENT
  #define LEN_INDENT                   3
  #define INDENT_WIDTH                 12
  #define BREAKSPACE                   " "
#else
  #define INDENT                       "\001"
  #define LEN_INDENT                   1
  #define INDENT_WIDTH                 (FW/2)
  #define BREAKSPACE                   "\036"
#endif

robustini, 2022-10-01:
Ok, I'll fix all the languages and create a PR.

raphaelcoeffic, 2022-10-01:
That'*s what I just saw, and makes no sense to me, so that I think we should just remove `BREAKSPACE` all together....

robustini, 2022-10-01:
> That'*s what I just saw, and makes no sense to me, so that I think we should just remove `BREAKSPACE` all together....

Ok, but in that case we would still have to create a condition because many of those sentences are too long for B&W displays.

raphaelcoeffic, 2022-10-01:
Just did this: https://github.com/EdgeTX/edgetx/pull/2432

Let's see if that's ok for everyone.

robustini, 2022-10-01:
> Just did this: #2432
> 
> Let's see if that's ok for everyone.

For me it is ok, but it will have to be checked on the TX with B&W if the whole sentence does not go beyond the screen.
So I would create some condition like (from english):

#define TR_DELETEMODEL                 TR ("DEL MDL" , "DELETE MODEL")

Or

#define TR_DELETEMODEL                 TR ("DELETE\nMODEL" , "DELETE MODEL")

If not necessary your draft is fine for me.

raphaelcoeffic, 2022-10-01:
This PR should not change anything to the current behaviour. If there are trouble with lines length, then yes, that `TR()` macro can be used. But if it wasn't necessary until now, why change it?

pfeerick, 2022-10-01:
Damnit @raphaelcoeffic ... let me finish writing my comment first! 😆 

Since that block is basically identical for all the languages - except one or two that aren't, so I think it block should be gotten rid of from each language and put somewhere more common... no need to repeat yourself twelve times (and get it wrong one or twice, as it looks to have been done for a couple of languages).

robustini, 2022-10-01:
LGTM

pfeerick, 2022-10-01:
Yup, especially when you look at the phrases affected - some were widescreen B&W only so didn't need to be wrapped, and others are small enough to fit on the not-so-widescreen B&W as it is (e.g. TR_NO_SCRIPTS_ON_SD fits on one line on T-Lite 128x64 screen) ... so it seems like a dubious way to do `TR()` without using `TR()` for multiple lengths (and pointlessly).

closed 2022-10-06 by pfeerick