EdgeTX/edgetx #2406: Allow Friendly Names for Logical Switches and Special Functions

Dweinbach, 2022-09-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I can't remember what a logical switch or special function was intended to do

### Describe the solution you'd like

I want to give each logical switch and special function a descriptive name. and then see those in places where they are used

this is similar to a secondary part of #1896 , BUT I have no need for the grouping, so I'm hoping this part will be easier/quicker to implement than the grouping described there

Eldenroot, 2022-09-28:
Good and useful suggestion! +1

pfeerick, 2022-09-29:
This will be implemented as part of https://github.com/EdgeTX/edgetx/pull/2085 and https://github.com/EdgeTX/edgetx/pull/2219

frankiearzu, 2023-10-10:
That will be great.. i have been trying to understand a complex plane template in SoarETX and having a name/comment for the Logical switches will help a lot.

trimore, 2024-01-19:
I would love this too.  Also, in addition to the name, would be great to have something like 50 to 100 characters to do a brief description or note.

Helipro, 2024-02-05:
This would be nice.  I use a a excel spreadsheet for many of my models.  Its the only way not to waste logical switches.  It helps tremendously when I have no logical switches or only a few left.  They can get really complex incorporating adjustable global variables using curves(for limits on a pan and tilt system) and combination of flight modes, and mixes to save logical switches.

teradactyal, 2025-03-20:
Looking forward to this being implemented as it get difficult to keep track of logical switches, even with only a few. Any news on when it might be incorporated?

JustPlaneChris, 2026-09-03:
Did this ever get any traction? Coming from Ethos, being able to name logic switches is sorely missed.

teradactyal, 2026-09-03:
I heard that in V3 this might be included.

prdeppe, 2026-11-08:
Named logical switches would be a huge benefit for the complex setups I use for gliders (i.e. E-Soar and similar) and for controlling ArduPilot.  Managing a complex set of interrelated logical switches requires maintaining separate documentation and ensuring that the switches and the documentation always match.  It's doable, but being able to name logical switches would save time and reduce the chance for errors.  I was hoping this would remain a 3.0 milestone.