EdgeTX/edgetx #2407: Allow the 6 buttons of "6P" on Radiomaster TX16S to be used as six individual Momentary Switches

Dweinbach, 2022-09-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I don't have enough switches, yet there are 6 switches staring me in the face

### Describe the solution you'd like

I'd like to see 6 additional Switches S1P-S6P(?), in the list of switches, so I can use them throughout in logical switches, special functions, etc

### Describe alternatives you've considered

1. I think this could be done by creating multiple logical switches, but that would be time consuming, and confusing

gagarinlg, 2022-09-28:
That is sadly impossible.
The 6 switches are connected to a small processor that controls them and the sends an analog voltage to the main processor, depending on the selected button.
EdgeTX has no control over the behavior of the buttons.

Eldenroot, 2022-09-28:
@gagarinlg - unfortunately... this could be nice addition, now the buttons have limited usage.

rotorman, 2022-09-28:
Well, the hardware does not allow it directly, but there is a software way of likely still achieving the desired momentary functionality. See this video by painless360: https://www.youtube.com/watch?v=c-eShYkcA7g

Dweinbach, 2022-09-28:
> Well, the hardware does not allow it directly, but there is a software way of likely still achieving the desired momentary functionality. See this video by painless360: https://www.youtube.com/watch?v=c-eShYkcA7g

Right - so my suggestion is to "hard-code" the 6 values and present them as switches, rather than consuming 6 logical switches and having to configure them for each model.

Eldenroot, 2022-09-28:
> Right - so my suggestion is to "hard-code" the 6 values and present them as switches, rather than consuming 6 logical switches and having to configure them for each model.

This would be awesome :)

pfeerick, 2022-09-29:
Um... they have been in the switches list for since at least ETX 2.5, so can be used for inputs, mixes and special functions. 😆 And also in LS with the "AND switch" test. I think if you haven't calibrated the 6POS yet, you'll only get 6P1 until you calibrate it properly. 

https://user-images.githubusercontent.com/5500713/192911511-d54be742-1f35-457a-83ce-b6ba96c22131.mp4

Dweinbach, 2022-09-29:
6P1 - 6P6 are basically what I was asking for, so closing this suggestion

closed 2022-09-29 by Dweinbach