EdgeTX/edgetx #2408: Duplication of models when changing models.     2.8

RipperGOT, 2022-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When you change the model and change again, the same duplicate model appears, if you repeat the step, it duplicates it again. Only when you reboot do the models reappear.

### Expected Behavior

Not duplicate models

### Steps To Reproduce

Select model

[image]

Change again the model
Duplicate model appears

[image]

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

dlktdr, 2022-09-29:
I can't seem to reproduce this one. Can you zip and upload your models and radio folder?

RipperGOT, 2022-09-29:
> I can't seem to reproduce this one. Can you zip and upload your models and radio folder?
Of course, here is!

[MODELS.zip](https://github.com/EdgeTX/edgetx/files/9675821/MODELS.zip)

dlktdr, 2022-09-29:
I see it shows that there is duplicated model1 (GE260Z) in labels.yml. As soon as I boot it clears it for me, and I still can't seem to repeat getting it to duplicate. Either in the simulator or on my TX16s. 

Can you send the radio.yml file too, maybe it's a combination of things. Seems like an important issue will try to get to the bottom of it.

Anyone else having this issue yet?

RipperGOT, 2022-09-29:
> I see it shows that there is duplicated model1 (GE260Z) in labels.yml. As soon as I boot it clears it for me, and I still can't seem to repeat getting it to duplicate. Either in the simulator or on [...]

Yes, I have seen it, but I am not sure if it was already duplicated or 2.8 did it
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/9678164/RADIO.zip)

RipperGOT, 2022-09-30:
> I see it shows that there is duplicated model1 (GE260Z) in labels.yml. As soon as I boot it clears it for me, and I still can't seem to repeat getting it to duplicate. Either in the simulator or on my TX16s.
> 

I have tried deleting labels.ylm, 2.8 re-creates the file without any duplicates. It is only when I change models several times that it duplicates them in the file.

dlktdr, 2022-09-30:
Thanks, that sounds like it might be some kind of a timing issue on the write of labels.yml then. Are you cycling back and forth between the same two models, or does it seem to matter?

Looks like we have another user who experienced the same. #2427

skunkwurk, 2022-10-01:
yes, It appears to occur when I cycle between the to models.

dlktdr, 2022-10-01:
Can someone who is getting this issue please test PR #2429, firmware should be available in actions shortly. 

I still have not been able to re-produce this personally.  I think I might have found a potential cause. Please let me know.

Thanks!

skunkwurk, 2022-10-01:
> Can someone who is getting this issue please test PR #2429, firmware should be available in actions shortly. https://github.com/EdgeTX/edgetx/suites/8562572182/artifacts/382692500
> 
> I still have not been able to re-produce this personally but think I might have found a potential cause. Please let me know.
> 
> Thanks!

I just tried it, it is no longer creating a duplicate.  Nice work!  I did lose 2 models though.  When the radio booted after I loaded the .bin, a pop-up displayed, "It is going to remove the unused models.  hit any key to continue". Two models I recently added are gone.  Do you need me to test anything else?  If not, I'm going to try and recover the 2 models.

dlktdr, 2022-10-01:
No. That's good to hear, I still wonder what is different from your radio to mine.. I've been trying for about 45min switching around and couldn't get it.

Just copy those models from /models/unused back into /models.. they should come back.

dlktdr, 2022-10-01:
@RipperGOT, can you confirm this as well?

skunkwurk, 2022-10-01:
I moved them from unused to model and they're back.  Is it ok to leave copies in the unused folder or should I delete them?  Also, is it ok/safe to run this PR code?

dlktdr, 2022-10-01:
> I moved them from unused to models and they're back. Is it ok to leave copies in the unused folder or should I delete them? 

Yeah doesn't hurt anything, up to you. Was an oversight, since it loads all models in the folder it brought back a lot of users deleted models see #2373. So this change (was added today) moves the unused ones here.

> Also, is it ok/safe to run this PR code?

It's the top of main + this change to the and pretty close to what is going to be RC2.. as with any release candidate or PR, I wouldn't be flying anything super expensive with it... but we would all like to hear any more major issues found. 😉

skunkwurk, 2022-10-01:
fair enough!

dlktdr, 2022-10-01:
> Wait, so keep the unused folder? Or ok to del that too?

It can be deleted too.

pfeerick, 2022-10-01:
I've been swapping models like crazy with another PR that's basically `main` and haven't been able to repo this duplication, so is truely a strange beastie. Hopefully https://github.com/EdgeTX/edgetx/pull/2429 resolves it for both as I certainly can't offer up a repo case atm

skunkwurk, 2022-10-01:
Morning. I have a follow up question. I see the nightly includes the following: Move mdls not in models.yml to /Models/Unused

I added two models while on 2.8.0 that was moved to unused.  Why would they not be included in models.yml?  What is the requisite, process or function which adds models to the models.yml file?

gagarinlg, 2022-10-01:
models.yml contained all available models
deleting a model did not delete the modelxx.yml file, but only removed the reference in models.yml, so that the model was not presented to the user anymore

skunkwurk, 2022-10-01:
> models.yml contained all available models deleting a model did not delete the modelxx.yml file, but only removed the reference in models.yml, so that the model was not presented to the user anymore

Got it and it makes sense.  And I believe this was added as a cleanup when legacy models began to reappeared when labels were introduced.  But my question is; why would two model I was actively using, not deleted, not be in models.yml?  It seems they should have been referenced.  I did not check prior to, so speculating.

gagarinlg, 2022-10-01:
When you could access the model with <2.8 they must have been in the models.yml.

skunkwurk, 2022-10-01:
The two in question I added while on 2.8.  That is  why I'm curious if there is something else happening in 2.8.  Anything that was existing from <2.8 did not get moved.  This is a new radio for me though, so I did not have many deleted models while on 2.7.1 to begin with.

gagarinlg, 2022-10-01:
With 2.8, since merging the model labels, models.yml is not used anymore. There might have been some corner cases, when you have already been on 2.8 with model labels, but before the fix, and then installed a version with the fix, that some models might have been moved away. That is the cost you pay, when using pre-releases or nightlies.

skunkwurk, 2022-10-01:
That is exactly what happened.  No worries, I enjoy the trouble and isolation aspect of this testing.  I appreciate the attentive support the community is putting into the development.

RipperGOT, 2022-10-01:
> @RipperGOT, can you confirm this as well?

Yes, I have tried it and it does not duplicate the models. No model has been lost.

gagarinlg, 2022-10-01:
> 

I appreciate and I am happy about the sudden rise of community engagement after the 2.8-RC1 release. Suddenly a lot more people started to help. Which is really great!

dlktdr, 2022-10-01:
I finally got this to happen on my other radio. It only did it on a model that had the yappu telemetery lua script running. Reboot refreshes back to normal.

This fix #2429 does work too, @pfeerick I think we should make sure this makes it into RC2. I can easily see why someone would delete (a good) model thinking they hit duplicate.

RipperGOT, 2022-10-01:
> I finally got this to happen on my other radio. It only did it on a model that had the yappu telemetery lua script running. Happens every time you open that model. Reboot refreshes back to normal.
> 
> This fix #2429 does work too, @pfeerick I think we should make sure this makes it into RC2. I can easily see why someone would delete (a good) model thinking they hit duplicate.

I have Yaapu telemetry active in almost all models.

Well seen!

pfeerick, 2022-10-01:
Fantastic., and thanks for all your help guys! :)

closed 2023-05-05 by gagarinlg