EdgeTX/edgetx #2409: Top Bar Widget disappears when you activate internal GPS.  2.8RC1

RipperGOT, 2022-09-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In 2.8, when you activate the internal gps and the icon appears next to the battery, the fourth widget disappears. In 2.7 it doesn't happen.

### Expected Behavior

The fourth widget should still appear

### Steps To Reproduce

Activate internal GPS

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

2.8

[image]

2.7

[image]

pfeerick, 2022-09-29:
It actually looks like the simulator and radio aren't updating the top bar properly when you turn the GPS option on. If you look closer in top bar settings mode at the size of the top bar widget frames, you'll notice the icon for the GPS actually overlaps the space for the fourth widget - and so the forth widget space is supposed to be turned off because of the collision. When you initially set it on the radio, it will show four top bar widgets, and then when you power cycle it, it will show three. Same would probably happen in the simulator if you changed modes, and it refreshed the top bar properly.

RipperGOT, 2022-09-29:
> It actually looks like the simulator and radio aren't updating the top bar properly when you turn the GPS option on. If you look closer in top bar settings mode at the size of the top bar widget [...]

Okay!!! I see what you mean! So, I guess there will be no solution, right?

pfeerick, 2022-09-30:
Just a bugfix so that the widget is properly cleared, but other than that, no, not at the moment I don't think. Top bar widgets are a constant size, if only for the sanity of the widget authors. And we can't move that GPS icon over further as there are actually other icons not visible in the "unused" space. If in the future we can make it so the icons/top-bar seconds can be turned on or off ... i.e. "who needs the time" ... well, that would resolve it... but that certainly won't be something added just atm.

RipperGOT, 2022-09-30:
OK, thanks mate!!! If you consider it appropriate, change the label to enhancement.

philmoz, 2023-10-10:
Is this still an issue now that all top bar elements are widgets (in 2.10)?

RipperGOT, 2023-10-12:
> Is this still an issue now that all top bar elements are widgets (in 2.10)?

I still have the same problem in 2.9.

3djc, 2023-10-12:
The question explicitly mentioned 2.10, not 2.9

pfeerick, 2024-03-04:
@RipperGOT would you be able to check this again with a nightly firmware (ensure you backup your RADIO/MODELS folders beforehand if you do)?