EdgeTX/edgetx #2412: Tab in Channel Selector is too small to acces with touch in 320x480 screen orientation

richardclli, 2022-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

1. Tab size too small
2. The chooser is align too left

### Expected Behavior

1. Enlarge the tab size for this screen DPI/orientation
2. At least center the chooser in the screen.

### Steps To Reproduce

1. Add a value widget
2. Tap the source
3. Can see the channel chooser popup

### Version

Nightly (Please give date/commit below)

### Transmitter

FlySky NV14

### Anything else?

[image]

pfeerick, 2022-09-29:
Doesn't even need to be the value widget.. just go in to model settings, mix, pick a mix line and hit source. For comparison, here's the TX16S:
[image]

richardclli, 2022-09-29:
Interesting that TX16S is centered align but EL18 is not.

Eldenroot, 2022-09-29:
This popup should be bigger, wasted space around

gagarinlg, 2022-09-29:
This is two pop ups draw next to each other. The first one is the input list and the second one the filter selector. The input list is centered and then the filter selector is drawn next to it

gagarinlg, 2022-09-30:
The pop up will be centered correctly soon, maybe the next nightly.
There is not much that we can do regarding the size of the filter buttons. We do not have those in a larger font currently. Also the next font step would be 24 instead of 16, which seems to be rather large for that application.

Eldenroot, 2022-09-30:
What about make a popup wider to make empty space smaller

Eldenroot, 2022-09-30:
Also one idea ia to move the category buttons on the top instead on the left, then you can use bigger icons

gagarinlg, 2022-09-30:
> What about make a popup wider to make empty space smaller

On the vertically oriented screens, there is not much empty space and if we make it wider on the horizontal screens, you would gain nothing, except having more white

> Also one idea ia to move the category buttons on the top instead on the left, then you can use bigger icons

They were on the left originally and I am not going to change that now.
The problem is the fimrware size increase when using bigger symbols. To have the symbols also in the 24pt font costs us about 1800 bytes of flash memory.
@raphaelcoeffic what do you think? It does look much better on the NV14 with the larger fonts, but...
[image]
[image]

gagarinlg, 2022-09-30:
[image]

I generally like the look with the larger symbols.

Eldenroot, 2022-09-30:
Bigger icons look nice! +1

pfeerick, 2022-10-01:
Bigger are better fore sure, and I don't think the increased flash usage is an issue.

raphaelcoeffic, 2022-10-01:
@gagarinlg you might think about adding some border pixels as well, which makes the contact surface bigger and less erratic with touch.

gagarinlg, 2022-10-01:
[image]
[image]

Like that?
Does anyone know what padding/margin that is on the right side, between the buttons and the right end of the menu?

pfeerick, 2022-10-01:
Is it possible to tweak a little bit more of a gap between the icons - i.e. a little vertical padding between each?

gagarinlg, 2022-10-01:
Just the filter icons or also the menu entries?

pfeerick, 2022-10-01:
I think the entry size is fine... it's the icons that are a bit... touchy ... but maybe it'll be ok with those larger icons anyway...

gagarinlg, 2022-10-01:
those are the larger icons

PR #2435 (2022-10-01): feat(color): Bigger menu toolbar icons, outlines around menu
chore: add EdgeTX source symbols to 24pt fonts to allow for larger symbols in menu filter toolbars

Fixes #2412 together with https://github.com/EdgeTX/libopenui/pull/70

[image]
[image]
closed 2022-10-01 by pfeerick