EdgeTX/edgetx #2413: Text Overlapped in Telemetry Screen

richardclli, 2022-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Some Text Overlapped, in both simulator and radio.
[image]
[image]

### Expected Behavior

Not overlap.

### Steps To Reproduce

Model Settings -> Telemetry page

### Version

Nightly (Please give date/commit below)

### Transmitter

FlySky NV14

eshifri, 2022-09-29:
It is a duplicate of #2254 
The problem exists for vertical screens only and is due to the fact that this screen was not converted to lvgl yet.

Eldenroot, 2022-09-29:
Maybe this should be fixed in 2.8 release? If not many new issues (duplicated) will pop up :(

pfeerick, 2022-09-30:
Duplicate of #2254

pfeerick, 2022-09-30:
@eshifri If you get a chance, can you see if blank a line or two can be inserted just for the portrait screens... i.e move sensors table down. I want to try and do some testing of two other PRs so they get merged in soon so may not have time to play with this as soon as I'd like.

closed 2022-09-30 by pfeerick

eshifri, 2022-09-30:
It is sufficient to use shorter text for vertical screen.
If you prefer an extra line, I can try it later today (hopefully).

pfeerick, 2022-09-30:
Are there other instances where shorter is better though... or is it basically just this instance? As if there are more I have no problem resurrecting that PR as a stopgap ;)

richardclli, 2022-09-30:
Strange that is LVGL should have better flow control to avoid overlapping, maybe this screen is not fully LVGL yet?

eshifri, 2022-09-30:
It is not lvgl-ed at all.