EdgeTX/edgetx #2414: Window::setTextFlags() doesn't support the font change

banyaszg, 2022-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Some older color UI code calls the Window::setTextFlags() to change the font type after the creation. The current LVGL based version doesn't support this. However it still saves the value to the Window::textFlags, but that value isn't used for displaying the labels or changing its properties.
The StaticText has an own implementation to use the textFlags, but that works only at construction time.

As I know the following parts are affected:
- Logical Switch view.
- About view
- Logical Switch edit page
- Special Function edit page

### Expected Behavior

We should fix the affected parts or create the support.

### Steps To Reproduce

grep -r "setTextFlags.*FONT" radio
:)

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

raphaelcoeffic, 2022-10-01:
This is legacy stuff that needs to be replaced / obsoleted. The only place where it will stay for a while is LUA (mapped to the proper interface bellow).

The proper way to fix this is to get rid of it ;-)

banyaszg, 2022-10-01:
Then I'm going fix the affected parts the same way as the Logical Switch monitor. (Except the About text, because that is static.)

PR #2440 (2022-10-02): fix(color): Remove rem. `Window::setTextFlags()` font change usage
Fixes #2414 

Summary of changes:
- About: The MessageDialog constructor was extended adding the text flags. So the settings could be added at construction.
- LS & SF settings: A custom state was added with the style settings.
closed 2022-10-02 by pfeerick