EdgeTX/edgetx #2415: No image on the screen splash

ThePhantom05, 2022-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When firmware with its own picture on the screensaver, the taranis x9 + transmitter via bootloader- the firmware does not start the transmitter does not start and there is nothing, it feels like he died.  If the original firmware with the native picture is all well stitched and the transmitter turns on. 
Please fix the problem, otherwise it seems to me the appendix is useless. One word dies maybe. They have lost the nightly versions and have not had an update for more than two months...And the coach doesn't work.

### Expected Behavior

I didn't expect this.

### Steps To Reproduce

1. I turn on the transmitter in bootloader mode. 2. I connect via USB. 3. I go to the EDGE TX companion. 4. I select all the parameters in the settings and my picture for the screensaver. 5 . I specify the firmware 2.7.1 (its location). 6. I click upload via companion - it is being stitched. 7. Exit to the bootloader, press exit in it, and nothing happens. The transmitter is off. the screen does not show anything.
P.S. After that it does not turn on at all until you turn off the battery. After that. Bootloader is loaded, you flash the original firmware without a picture and everything works.

### Version

2.7.1

### Transmitter

FrSky X9D+

pfeerick, 2022-09-30:
It sounds like you are facing a slightly different issue - i.e. I don't think it's specific to adding a custom splash screen to the firmware as I have seen this with non-custom firmware. We are trying to identify/find some common cause for when some firmware updates via the bootloader do not seem to work properly for some B&W radios - X9D+, in particular, seems to be guilty of this. However, I believe a DFU update of the exact same firmware  *does* work.

ThePhantom05, 2022-09-30:
Thank you for the answer) I didn't try it through DFU (through the firmware) only through the companion. It was late yesterday. I found a topic by chance in the Edge tx Flasher topic about this error - it has not been fixed yet. 
P.S. Open TX and did not answer. Thank you for the answer. I will wait for the correction and return to this firmware.

ThePhantom05, 2022-09-30:
Thank you for the answer) I didn't try it through DFU (through the firmware) only through the companion. It was late yesterday. I found a topic by chance in the Edge tx Flasher topic about this error - it has not been fixed yet. 
P.S. Open%

pfeerick, 2023-05-07:
The 
> We are trying to identify/find some common cause for when some firmware updates via the bootloader do not seem to work properly for some B&W radios - X9D+, 

part has been pinned down well and truely... it was the undetected flash overflow. It is most likely this was related to that, and as of 2.8.4 should no longer be an issue 🤞

closed 2023-05-16 by pfeerick

epareja2, 2023-06-18:
Hello, I just want to let you know that I installed version 2.8.4 on a Taranis 
X9d+ and I still cannot load the splash screen.

I tried to upload the firmware with DFU and with normal SD USB connection but I always get the EDGE TX splash screen.

[image]