EdgeTX/edgetx #2416: Failsafe UI doesn't allow copying of channel value to failsafe for individual channels

RC-SOAR, 2022-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Running 2.8.0 RC-1 on TX16S

In OTX, when using the MPM with FrSky and FlySky protocols, there's a way to copy _individual_ channel values to failsafe. However this option is lacking in EdgeTX (you can only copy _all_ channels to failsafe, as in OTX).

In OTX, the interface works as follows (non-touch, TX16S). 
1. Go to Model Setup menu
2. Select Failsafe mode = custom
3. Choose 'set'. Takes you to the failsafe menu with all channels displayed
4. Highlight the channel 
5. Click {ok} to place in edit mode
6. Long press {ok}. The channel value is copied, with beep to confirm. Alternatively, you can set the value manually.

The Zorro / OTX has the option to 'Channel=>Failsafe' in a dropdown (other options include 'None', 'Hold', 'Channel**_s_**=>Failsafe')

 

### Expected Behavior

See above

### Steps To Reproduce

See above

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

TX16S / 2.8.0 RC-1

raphaelcoeffic, 2022-10-01:
@RC-SOAR what about having a second button copying the current channel value?

[image]

I'd have to find a suitable icon in those already in the font.

pfeerick, 2022-10-01:
Have a look at #2417 also if you're fiddling with the failsafe page ;)

RC-SOAR, 2022-10-01:
@raphaelcoeffic Yes, that would work. 
Another suggestion: is just have it as an additional action when you hit the gear icon - a tap would immediately copy the channel value into the failsafe value field.

closed 2022-10-11 by pfeerick