EdgeTX/edgetx #2417: No acceleration (or acceleration very weak) when changing failsafe values with roller.

RC-SOAR, 2022-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Running 2.8.0 RC-1 on TX16S.

In the failsafe menu it's possible to alter a channel's failsafe value using the roller, however the click increment is only 0.1, and the acceleration is very weak. This makes it very slow go from one end of the range to the other.

### Expected Behavior

It should be possible to span the range -100 to +100 in a few seconds.

### Steps To Reproduce

1. In MPM options, choose FrSky D16 protocol
2. Open the Failsafe menu
3. Use the roller to change a failsafe value

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

TX16S / 2.8.0 RC-1

pfeerick, 2022-09-30:
Ok, so need to check if #2356 needs extending to another component.

raphaelcoeffic, 2022-10-16:
The basic issue here is that the value range is quite different.

Most values have a range between `-100` and `+100`, while these limits have a range from `-1024` to `+1024`. So we would need to detect this and apply some multiplication factor.

PR #2584 (2022-10-19): Improve rotary encoder acceleration
Fixes #2417

Requires EdgeTX/libopenui#78

Summary of changes:
- use a more accurate / smoother implementation
- added specific acceleration factor which can be tuned for each number edit (depends on range / usage)
commit ba713b6b (2022-10-19): feat: Further improve rotary encoder acceleration (#2584)

* Improve rotary encoder acceleration

Fixes #2417

* Further improvement

* Fix simu compilation
commit 0c55c004 (2022-10-31): feat: Further improve rotary encoder acceleration (#2584)

* Improve rotary encoder acceleration

Fixes #2417

* Further improvement

* Fix simu compilation
closed 2022-10-19 by pfeerick