EdgeTX/edgetx #2420: Swedish sound files can not be downloaded

ulfhedlund, 2022-09-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When trying to download/update Swedish sound files I get an error "Language se not listed in sounds.json" and the download exits with an error.

### Expected Behavior

The sound files can be downloaded

### Steps To Reproduce

1. select "se" language
2. set sounds to be updated
3. run the update

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.8 RC1

[image]

elecpower, 2022-09-29:
The issue is due to the radio using 'se' as the language and implemented in 'radio/src/translations/se.h' and therefore Companion tries to find entries using the same prefix in the sounds.json file and cannot find any. The file contains this entry:
"language": "sv-SE",
"name": "Swedish Female",
"description": "Swedish Female Voice (sv-SE-SofieNeural)",
"directory": "sv"

@ulfhedlund do you have to rename the 'sv' directory to 'se' on the radio sd card for the translations to work?

elecpower, 2022-09-29:
[image]

elecpower, 2022-09-29:
<html>
<body>

sv | Swedish
-- | --
sv-FI | Swedish (Finland)
sv-SE | Swedish (Sweden)

</body>
</html>

pfeerick, 2022-09-29:
If I understand this right, it will only require a rename on one side or the other - sound pack or firmware. We *could* do the firmware, but to my mind that is a bit weird - sv instead of se for Swedish, zh instead of cn for Chinese. IMO the path of least resistance is just to use the names Companion/Firmware is looking for for now for the voice pack, and once we have the firmware a bit more flexible about what voice packs it can handle, we can start doing the ISO naming. But I'm cool with what happens either way - done in firmware/companion, done in the sound pack.

ulfhedlund, 2022-09-30:
> The issue is due to the radio using 'se' as the language and implemented in 'radio/src/translations/se.h' and therefore Companion tries to find entries using the same prefix in the sounds.json file [...]

I haven't tried this yet. Will do over the weekend and report back.

ulfhedlund, 2022-10-01:
In radio: If I rename the directory from 'sv' to 'se' on the SD card I get Swedish sounds.
In Companion: I can't seem to get the Swedish sounds to play at all, regardless of the directory being named 'sv' or 'se'...

elecpower, 2022-10-02:
Thankyou for checking radio. You proved what I expected that we need to align sounds directory too in the json file.
Companion simulator will also need the directory renamed in the sd structure in the radio profile.

elecpower, 2022-10-02:
Thankyou for checking. The rename confirmed what I thought. Means we have to also change the directory setting in the json file.
I'll do some testing of the sound files from SFs.

HThuren, 2022-10-20:
Hi @elecpower , @ulfhedlund and @pfeerick 
What are the right solution, I have the same issue with Danish sounds, catalog named dk, but expected are "da" .
Where to solve, firmware or rename i sounds.json ?

HThuren, 2022-10-20:
Ps update of danish files in catalog "dk", was nicely found by new automatic update feature.

HThuren, 2022-10-23:
well, changed the subcatalog to 'da', hope this will do :-)

MarSik, 2023-01-21:
The same issue happens for Czech sounds as well. The companion looks for cz, but the sounds.json file uses cs_CZ.

HThuren, 2023-01-21:
> The same issue happens for Czech sounds as well. The companion looks for cz, but the sounds.json file uses cs_CZ.

I think the best are simply rename, as we did for dk/da sv/se, seems to be first part xx_YY, ie. xx are used to find sound map.

pfeerick, 2024-03-05:
@HThuren @ulfhedlund  What's the current status of this issue? I seem to vagley remember renaming stuff on the sdcard-sounds repo but don't want to close this before confirmation that it was actually resolved! :grinning:

ulfhedlund, 2024-03-06:
@pfeerick Haven't been using Swedish sounds myself in quite a while. I believe it is fixed but I can not check until Sunday at the earliest, 
I am on a skiing trip in the Swedish mountains :)

ulfhedlund, 2024-03-09:
@pfeerick Just got back and successfully downloaded Swedish sounds from Linux Companion 2.10 rc1. I.e. it works as expected and the sound files are correctly stored in the SOUNDS/se directory.