EdgeTX/edgetx #2425: Buttonsupport in Keyboard for non Touchradios

ParkerEde, 2022-09-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The new keyboard in 2.8.0 has become much more cumbersome for radios without a touch function than it was before. To allow a little easier input, I would like to propose something.

### Describe the solution you'd like

The buttons SYS, MDL, TELE and PgUp/Dn are currently without function. How about assigning the following functions to these buttons:
SYS = Arrow left
TELE = Arrow right
MDL = toggle CapsLock activ/inactiv

Horus X10
PgUp/Dn = toggle marked letter uppercase/lowercase

Horus X12
PgUp = marked letters to uppercase
PgDn = marked letter to lowercase

Eldenroot, 2022-09-30:
Great idea!

ParkerEde, 2023-11-22:
@philmoz what do you think about this?

philmoz, 2023-11-22:
I've added PR #4355 for this.

Please see the PR for the shortcut mappings I have chosen (different to what was suggested here).

ParkerEde, 2023-11-23:
Basically, the shortcuts work. However, the keys/buttons are different from those in the description above. Would you please use the keys according to this list?
SYS = Arrow left (not Backspace)
TELE = Arrow right
MDL = toggle CapsLock activ/inactiv

Horus X10:
PgUp/Dn = toggle marked letter uppercase/lowercase

Horus X12:
PgUp = marked letter to uppercase
PgDn = marked letter to lowercase

philmoz, 2023-11-23:
I chose the shortcuts for the following reasons:
- The PG keys are normally associated with movement so it made sense to have them move the cursor.
- Using two buttons to change case is redundant, you only need a single button to flip the case. I chose TELE.
- There are four keyboard layout, not just upper and lower alpha. SYS made sense to do this (changing keyboard SYStem)
- That left the MDL button for backspace/delete.

ParkerEde, 2023-11-23:
My thoughts were that these shortcuts would primarily benefit the non-touch radios. Hence this wish
[image]
In my opinion, this is the most intuitive way

philmoz, 2023-11-23:
> My thoughts were that these shortcuts would primarily benefit the non-touch radios. In my opinion, this is the most intuitive way

Thanks, I understand now (not having any of these radios makes is harder to visualise when I'm madly coding away).

This is what I'm thinking:
- Radios with only a single PGUP/DN key - your mapping as shown in the image above, plus LONG MDL = backspace and LONG PGDN = Delete
- X12S - as above; but PGUP also toggles case
- Radios with both PGUP and PGDN (except X12S) - mapping as per PR.

philmoz, 2023-11-23:
For the number keyboard.

Radios with a single PGUP/DN button and X12S:
- SYS = '-'
- LONG SYS = 'MIN'
- TELE = '+'
- LONG TELE = 'MAX'
- MDL = '<<'
- LONG MDL = '+/-'
- PGUP/DN = '>>'
- LONG PGUP/DN = 'DEF'

Radios with separate PGUP/PGN buttons:
- SYS = '<<'
- LONG SYS = 'MIN'
- MDL = '>>'
- LONG MDL = 'MAX'
- PGDN = '+'
- PGUP = '-'
- TELE = '+/-'
- LONG TELE = 'DEF'

pfeerick, 2023-11-23:
Just for reference, the colorlcd radios without touch are
- FrSky X10
- FrSky X12
- Jumper T16/T18
- RM TX16S (early)

So the physical button layouts could be

X10
[image]

X12
[image]

T16 (T18 is effectively the same)
[image]

TX16
[image]

So while this is layout is optimised for the X10, should work just fine for the X12, and be quite usable on the T16 & TX16

philmoz, 2023-11-23:
I've updated the PR with the alternate key mappings (current mappings are in the PR).

ParkerEde, 2023-11-24:
Tested with X10SExpress,RMTX16 and T18. Good job. Thx

PR #4355 (2023-11-27): feat(color): Shortcut keys for text keyboard.
Fixes #2425
Resolves #265

Adds shortcuts to the text keyboard using the hardware keys.

Short press:
- PG>/PGDN - move cursor right
- PG</PGUP - move cursor left
- SYS - cycle through keyboard layouts (alpha uppercase, alpha lowercase, numbers & symbols, numeric)
- TELE - change case of selected character (upper to lower or lower to upper)
- MDL - delete character to the left of the cursor (backspace)

Long press:
- PG>/PGDN - move cursor to the end of the input
- PG</PGUP - move cursor to the start of the input
- MDL - delete currently selected c [... cut]
closed 2023-11-27 by pfeerick