EdgeTX/edgetx #2427: Duplicate model being created as an unlabeled model

skunkwurk, 2022-09-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Good Morning.  I've been testing this RC and ran into a situation where the current model appears as a new model under but is unlabeled.  When I noticed the behavior I simply deleted the duplicate.  After I rebooted I noticed the model must have been referencing the same model ID, as the both the model I deleted and the labeled model had been deleted.  

### Expected Behavior

Model labels should not change without intervention.

### Steps To Reproduce

Once I recovered the model from backup, I was able to reproduce the behavior by adding a new model and then switching between models.  This seems to be the trigger.  This is only true until you reboot.  For example,  if you notice the duplicate and delete it, it will be gone when you reboot.  If you do not delete it and reboot, there appears to be a read of the yaml files that occurs.  When this takes place, it clears the duplicate and the radio appears to behave correctly. 

I hope I explained this clearly. I am open to questions if my explanation is unclear.   Thanks in advance!

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

dlktdr, 2022-09-30:
This sounds like a duplicate of #2408. Where the switching between models causing the issue. Can you take a video of the actions needed to make this happen. and post it in #2408. I haven't been able to make it happen myself yet or I'm missing something, along with any other details.

It looks like its not actually making a duplicate, but instead renaming another model in memory, on the reboot&rescan, all goes back to normal (if didn't delete in the meantime.... )

skunkwurk, 2022-09-30:
I tried to upload the video to #2408, but the file limit is 10mb.

pfeerick, 2022-10-01:
You can upload to a video share site like streamable or as unlisted video on youtube and share the link - directly adding video can sometimes be a pain here.

pfeerick, 2022-10-01:
Duplicate of #2408

closed 2022-10-01 by pfeerick