EdgeTX/edgetx #2436: 2.8RC1 BLE Joystick It stops working when you do a long press to exit.

RipperGOT, 2022-10-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In 2.7 when you activate the BLE joystick you can do a long press to exit lua and it stays active. In 2.8 if you long press to exit, it stops working. 
Sometimes I get this error: Script syntax error ?:0: attempt to index field '?' (to nil value) 
I also see the Lua a little lazy

### Expected Behavior

You could long press RTN to exit the LUA and keep it active.

### Steps To Reproduce

1 Enter in ELRS LUA-BLE Joystick and activate
2 Long press RTN to exit to model screen
3 BLE stop working

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[image]

RipperGOT, 2022-10-13:
In RC3 the ELRS lua is still lazy and the BLE joystick crashes from time to time. I also can't get out of the BLE without it disconnecting.

pfeerick, 2022-10-14:
By lazy, do you mean slow? As I've noticed that also, and it's not LVGL specific as B&W also seems sluggish at times.

pfeerick, 2022-10-14:
PK from ELRS just commented that we must be sending the `EVT_VIRTUAL_EXIT` event on exit (which I think we are), whereas 2.7 just killed the script. So somehow that is probably triggering the stopping of wifi/ble that would happen on a short RTN press.

RipperGOT, 2022-10-15:
Exactly, it's slow. It looks like lag.

raphaelcoeffic, 2022-10-16:
> Exactly, it's slow. It looks like lag.

This is caused by the fact that we switched to LVGL, and the LUA script does a full redraw on each frame. It is faster on radios that do not require the screen buffer to be inverted in software (T18, NV14, EL18). The screen inversion (180° rotation) in software costs us around 20ms on each frame. That does add up to the rest, which makes it look slower.

raphaelcoeffic, 2022-10-16:
> PK from ELRS just commented that we must be sending the `EVT_VIRTUAL_EXIT` event on exit (which I think we are), whereas 2.7 just killed the script. So somehow that is probably triggering the stopping of wifi/ble that would happen on a short RTN press.

This is probably caused by the fact the event still gets passed to the script while we actually kill it. I can check if there isn't some weirdness here.

gagarinlg, 2022-10-16:
Long term goal: expose the UI API to Lua in a way, that LVGL handles partialnredraws of Lua scripts

PR #2573 (2022-10-17): fix(lua): send EVT_VIRTUAL_EXIT only on short EXIT press
Fixes #2436

Summary of changes:
- `EVT_KEY_BREAK(KEY_EXIT)` is now mapped to `LV_EVENT_CANCEL` (previously `EVT_KEY_FIRST(KEY_EXIT)`)
- `short` and `long` `EXIT`/`RTN` work the same everywhere (`BREAK` event is generated, thus `onCancel()` is called)
- except in LUA where `long EXIT` **does not** generate a `BREAK` event.
commit 9f48769b (2022-10-18): fix(lua): send EVT_VIRTUAL_EXIT only on short EXIT press (#2573)

Fixes #2436
commit 4c1cebe0 (2022-11-06): fix(lua): send EVT_VIRTUAL_EXIT only on short EXIT press (#2573)

Fixes #2436
closed 2022-10-17 by pfeerick