EdgeTX/edgetx #2442: Portrait screen - spacing and update tweaks

pfeerick, 2022-10-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

1. The buttons on the Model Setup tab are not as nicely shown on the NV14 as the TX16. 
[image]

2. The outputs page has some visual quirks, which seems to go away as soon as you scroll... btw, when I tried to screenshot it on hw, the screenshot showed what it *should* look like... but simu did capture it properly. And it's not theme related, the same happens with the default theme. 
[image]
[image]

### Expected Behavior

1. I'm in two minds about whether it just needs a little more padding between each the first two and the third button of each group, or if it should be put in a grid, and the grid changed depending on screen dimensions (i.e. like the flight mode tab). 

2. That it look the same as it does after scrolling when you first enter the page. 

### Steps To Reproduce

1. Go into the model setting page

2. Go to the outputs tab, inspect and then scroll a bit. 

### Version

Nightly (Please give date/commit below)

### Transmitter

Flysky NV14

banyaszg, 2022-10-06:
The channel monitor is also not centered, and there are big gaps. Maybe a one column version would be better.
[image]

pfeerick, 2022-10-06:
I actually think it would be better to increase the number of channels shown to 16, and reduce the number of tabs, but sure will look at single column as would be easier to implement.

pfeerick, 2022-10-09:
> The channel monitor is also not centered, and there are big gaps. Maybe a one column version would be better. ![image]

Just looking at this - the off-centering is deliberate - there are icons shown on the left side for reversing and channel override... it looks like they got pushed too far to the left though at some point though 😮 

[image]

raphaelcoeffic, 2022-10-12:
> there are icons shown on the left side for reversing and channel override...

These icons really don't look good. Maybe we could just use another color for "override" channels? (`WARNING` red?).
For the channel reverse, frankly, I don't think you need anything: you just see that it moves the opposite way...

pfeerick, 2022-10-12:
Yeah, there are probably better ways to do it. The icons *are* aligned now though... I couldn't leave them where they were since I was fiddling around there. 😆

richardclli, 2022-10-18:
The file listing did not make use of full screen.

[image]

eshifri, 2022-11-01:
The area in the bottom is used to display model images, if you go into MODELS folder. This is similar to the empty right side for horizontal screen.

pfeerick, 2023-03-22:
Button alignment was fixed in a 2.9 PR

PR #2489 (2022-10-09): fix(color): outputs in portrait mode
Fixes #2442

TODOs:
- [ ] padding between buttons on *model setup* screen.
commit a8f19735 (2022-10-09): fix(color): outputs in portrait mode (#2489)

Fixes #2442
commit 714b102d (2022-10-10): fix(color): outputs in portrait mode (#2489)

Fixes #2442
closed 2023-03-22 by pfeerick